cameroncros/OctoPrint-DiscordRemote #104: Problem with configuration

megarvey47, 2019-09-07:
Hello. Been trying to get this plugin working for a bit now, and I have a feeling I'm looking right at the problem but can't seem to put my finger on it. 

Every time I try to save configuration settings the octoprint server becomes unresponsive. I thought maybe I had some bad input so I rebooted and realized I had mistyped the channel ID, so I corrected that and tried again, but the same result. I grabbed the log to see if I could ascertain the problem from the error output. The error message read "Missing Access", but I wasn't exactly sure what that implied.

I also noticed when I rebooted and opened up the DiscordRemote settings window again that my old incorrect info was prefilled in the boxes. So I thought maybe that was the problem, I tried editing the info once more to the correct ID values but the same issue. I tried uninstalling and reinstalling the plugin, but the incorrect ID info was still prefilled when I went to configure again. I'm not sure if this is a separate problem or the source of the problem. 

I've attached a snippet of the log file with the error message. Any help you can provide me is greatly appreciated.
[image]

cameroncros, 2019-09-07:
The slow/unresponsive saving is something I am loosely aware of, but I don't think it's a problem.

Have you definitely configured your not correctly and allowed it access to your server and channel?

cameroncros, 2019-09-08:
@megarvey47, have you solved your problem or is it still not working?

megarvey47, 2019-09-08:
Still not working. I was researching bot permissions and believe I have it setup correctly. I followed the instructions from the "How to get a token and channel ID from Discord" linked in this plugin's setup instructions, including the part where you add the bot to your channel with read and write messaging permissions. As far as I can tell the bot is on the server and in the channel, as it displayed "A wild octobot has appeared" text similar to when a user joins. I even tried explicitly giving the bot permission on the server to access messages, but no change.

megarvey47, 2019-09-08:
Just now trying again to configure settings, I only filled in the information I am 100% sure to be correct (i.e. the bot token and the channel ID) and nothing else, a new error in the log appeared. Similar to the first, except instead of "Missing Access" it is returning "Missing Permissions", so I must have misconfigured permissions, though I'm not sure where I went wrong.

cameroncros, 2019-09-08:
Have you allowed your bot to create and send messages to your discord servers channel? That would be the next thing to look at. 
Does the little green circle show up next to the bots icon?

megarvey47, 2019-09-08:
Yes, the bot is allowed to send and receive messages. I kicked the bot from the channel and re-added it just to be sure, but the little circle next to the bot's icon is gray stating it's offline.

cameroncros, 2019-09-09:
What permissions did you give the bot in the channel? It should have read and write messages, not really sure if it needs more.

megarvey47, 2019-09-09:
I went back and gave the bot all the text permissions and tried configuring again. The octoprint server no longer endlessly "saves" rendering the server unresponsive, now it accepts the settings as it should. Now checking the bot on discord, I see that it now shows the green circle indicating it's online, but I didn't receive a message on the discord channel and typing /help or /status doesn't return a response so I checked the log again and we're back to the "Missing Access" message, accompanied by a new error which I've attached a image of:

[image]

You'll notice the familiar "Missing Access" responses sandwiching a traceback reporting "TyperError: expecting string or buffer". The plot thickens.

cameroncros, 2019-09-09:
That sounds like your getting a lot closer, having the green light is a good sign. The traceback there is interesting, I'm not sure how that is being triggered. 

The missing access thing still seems to be a discord permissions thing, according to: https://www.reddit.com/r/discordapp/comments/7b1pwe/discord_bot_missing_access/

Try clicking the "DiscordRemote" icon at the topbar of the Octoprint UI (I think it looks like a game controller?) and it should prompt you to send a command. Try sending something and seeing if that helps? Otherwise I would consider giving more permissions?

megarvey47, 2019-09-09:
Alright I'm happy to report that after giving the bot Administrator privileges a wave of welcome messages came through on the server, awesome! By clicking the status button at the top of the octoprint UI, I get the expected message in my discord channel. I can communicate back and forth and am able to send commands as I expected. Thank you so much for all your help, and great work on the plugin by the way!

closed 2019-09-09 by megarvey47