cameroncros/OctoPrint-DiscordRemote #107: Discord Bots offline / not responding to commands.

Magicanz, 2019-09-25:
I've got 5 printers connected to one octopi, 3 of them have got cameras. The bots for those three often do not work. At a few times they start working one by one but quickly stop once again. The log files keep complaining about discordapp 502 bad gateway. I've attached the octopi log file. I'm not very skilled in programming, is this something I can fix or is it a discord problem. Is there a way to automatically restart the bot after these errors? Thanks!

[octoprint (1).log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/3653088/octoprint.1.log)

cameroncros, 2019-09-27:
This is a very interesting failure, i have no idea what could be causing it.

cameroncros, 2019-09-30:
@Magicanz, could you give me some more information about how you setup your 5 servers on 1 pi? I'm very curious to know how to do it so I can try replicate the issue.

On review of the code, I have noticed some minor issues that may be causing your problem, I will fix and push them in the coming days

Magicanz, 2019-10-01:
@cameroncros It was some while ago now I set the Pi up, but this was the guide I used: http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers. I started out with three cameraless printers, put cameras on them within a month, then installed 2 new printers a couple months later.

cameroncros, 2019-10-04:
Please upgrade to 2.51, and see if your problem still exists.

cameroncros, 2019-10-04:
And if you can give any more log files, that would be greatly appreciated.

Magicanz, 2019-10-09:
Problem seems to be fixed! Have only tried for about an hour though. Do you still want logs? (Sorry for late reply)

cameroncros, 2019-10-09:
If anything goes wrong, reopen and add logs, otherwise I'm glad it's fixed :)

closed 2019-10-09 by cameroncros