cameroncros/OctoPrint-DiscordRemote #109: /outputstatus seems to be broken?

dsdanielko, 2019-10-23:
The discord bot does not respond with anything, otherwise, /outputon and /outputoff works perfectly.

cameroncros, 2019-10-24:
Could you attach your log messages? I'll look into this soonish

dsdanielko, 2019-10-24:
how do i see the log sorry

cameroncros, 2019-10-24:
https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

dsdanielko, 2019-10-24:
There is nothing in the plugin_discordremote.log for either /outputon or /outputstatus but there is an error in octoprint.log around the times when I tried /outputstatus

> 2019-10-24 11:57:53,732 - websocket - ERROR - error from callback <bound method Discord.on_message of <octoprint_discordremote.discord.Discord instance at 0x75cc1df0>>: object of type 'bool' has no len()

[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/3767149/plugin_discordremote.log)

closed 2021-10-17 by cameroncros