cameroncros/OctoPrint-DiscordRemote #113: Plugin drastically slows down system.  Had to uninstall.

evannadeau, 2020-01-29:
I was having interface performance, as well as reboot issues after this plugin was installed.  Very sadly, I found it to be this plugin, but I truly want to use it.  Are there logs, or anything else I can provide to diagnose what could be causing this?

OctoPi 0.17.0
OctoPrint 1.3.12
Raspberry Pi 4 B.

cameroncros, 2020-01-30:
@evannadeau 

Thats definitely unusual, if you look in the logging area on the octoprint webui, there should be logs for both octoprint and DiscordRemote. Can you upload both and I will investigate if there are any obvious causes.
I personally run DiscordRemote on a PI3, so i am very suprised you are getting noticable performance/reboot issues.

evannadeau, 2020-02-02:
After a bunch of testing on a fresh deployment of OctoPrint, I'll say that the only real issue is how this plugin affects the stopping of the OctoPrint daemon.  This is noticed when performing any reboot or shutdowns of the system.  See the attached picture showing that as soon as this plugin is installed and enabled, stopping the OctoPrint daemon takes about 30 seconds.

[image]

cameroncros, 2020-02-02:
Hmm, that is true, and always has been. I had to write the discord bot implementation myself, and it is far from perfect. When octoprint finally leaves python 2, I will switch the plugin to use discord.py. In the meantime, i might try look into the shutdown issue if I get a chance, but I can't promise anything

evannadeau, 2020-02-02:
Fair enough, I'll close this issue.  I am now testing with OctoPrint 1.4.0rc4, which is Python 3.  If you would like someone to help test, please let me know.

closed 2020-02-02 by evannadeau