cameroncros/OctoPrint-DiscordRemote #114: In Octoprint 1.4.0 print start events are not sending

dshokouhi, 2020-03-07:
I just updated to Octoprint 1.4.0 and I noticed that the print started messages are not coming through in discord.  I think maybe something changed?

I also noticed that upon a restart if my printer is offline I get the printer in error state message.

Thanks for the awesome plugin btw!!

cameroncros, 2020-03-14:
Oh, sorry for not noticing this earlier. I will try investigate this as soon as I can. Do you have any log messages that would help narrow down the issue?

dshokouhi, 2020-03-14:
Ah sorry should've included that :) 

```
2020-03-13 18:51:54,045 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-03-13 18:51:54,064 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Stormfly.gcode, owner: boogie, user: boogie
2020-03-13 18:51:54,168 - octoprint.plugins.DisplayLayerProgress - INFO - Printing started. Detailed progress started.{'origin': u'local', 'name': u'Stormfly.gcode', u'user': 'USER', u'owner': 'OWNER', 'path': u'Stormfly.gcode', 'size': 45715512L}
2020-03-13 18:51:54,212 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-03-13 18:51:54,244 - octoprint.plugins.discordremote - INFO - Received event: printing_progress
2020-03-13 18:51:54,265 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-03-13 18:51:54,945 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': u'local', u'throttled_now': False, u'throttled_mask': 327680, 'file': '316fbfc68db5557a56b0649ea33d33cf7ab0d42a', u'throttled_past': True}
2020-03-13 18:52:00,834 - octoprint.plugins.discordremote - INFO - Received event: printing_started
2020-03-13 18:52:00,985 - octoprint.plugin - ERROR - Error while calling plugin discordremote
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_discordremote/__init__.py", line 280, in on_event
    return self.notify_event("printing_started", payload)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_discordremote/__init__.py", line 408, in notify_event
    return self.send_message(event_id, tmp_config["message"].format(**data), tmp_config["with_snapshot"])
KeyError: 'file'

```

cameroncros, 2020-03-15:
Just checking, are you running on python2 or python3?

dshokouhi, 2020-03-15:
python 2

cameroncros, 2020-03-15:
Easy fix:
[image]
Change {file} to {path}. I will update the defaults

dshokouhi, 2020-03-15:
Thank you 🙏 I will leave it to you if you want to close this now or not but I will change it on my end.

commit e0e4ee66 (2020-03-15): Fix #114: Incorrect key to get file path
closed 2020-03-15 by cameroncros