cameroncros/OctoPrint-DiscordRemote #115: Percentage reporting doesn't match what is shown in OctoPrint web GUI

Fenrirthviti, 2020-03-18:
Currently, I'm using the PrintTimeGenius plugin, which I think is affecting how DiscordRemote reports completion. It reports improper current % completion and sends status messages that don't match the actual progress. I am unsure what information I could provide that might help troubleshoot this, but please let me know if I can provide any debug information or other logs.

I see a previous issue opened a few years ago, but can't find any follow up on that. Is this still something I should report to the PTG folks to look in to? Is there a specific function/method being used by DiscordRemote to get the percentage that they should be aware of I can provide to help narrow things down? They might be overriding the GUI, but not a backend function that plugins can use, so it might be a simple fix?

cameroncros, 2020-10-18:
This is still a PTG/octoprint problem. I cannot help anymore unfortunately,

closed 2020-10-18 by cameroncros

Fenrirthviti, 2020-10-18:
Any additional information I can provide to the PTG team that might help them correct the issue?

cameroncros, 2020-10-19:
Just mention that other plugins that report print time have the old print time. You can tag me on the issue and I'll give any information they ask for