cameroncros/OctoPrint-DiscordRemote #116: [Feature Request] Audio signal with messages

Arkonos, 2020-03-20:
I'd like to get an audio notification when the printer is done or encountered an error.
Per #83 mentions don't work for that. I was also unable to trigger the [better-airhorn bot](https://top.gg/bot/503996428042108928) by including its command in the message.
I tried to use the [M117 Speech Synthesis Settings](https://plugins.octoprint.org/plugins/M117SpeechSynthesis/) plugin, but it doesn't work for me. If your plugin would support my wish I'd not need it however.
Thank you very much in any case

cameroncros, 2020-03-20:
What do you want to generate the noise, the printer itself or discord?
Currently, DiscordRemote only interacts with the text channels, it has no ability to generate sound effects. I am not sure I want to expand the functionality of DiscordRemote to send sound effects, as it would be a fairly significant undertaking, and I dont have much time to spend on this plugin as is. However, if you wish to make the change yourself, I will be happy to merge it. I can assist where possible as well.

Arkonos, 2020-03-20:
I'd like discord to make a noise, but have since read that most bots don't react to other bots.
Totally understandable that you don't want to create more work for you, thank you nonetheless.

closed 2020-03-20 by Arkonos