cameroncros/OctoPrint-DiscordRemote #117: Ability to @ users and roles not working

IanJDan, 2020-04-09:
I've been trying to @ users and roles within the various configurable messages and it doesn't seem to work when entering any of the following values:
@UserName
@UserID
@UserNickname
@RoleName
@RoleID

I would really only expect the User/Role IDs above to work, but none work currently. I'd appreciate any assistance with this so I can alert a role when a job has completed.

cameroncros, 2020-04-10:
See #83. Unless something has changed, I dont beleive that bots can @mention people in embeds. It might be possible with some effort though, as https://github.com/discordjs/discord.js/issues/1519 has a solution for discord.js. I am fairly busy with other tasks atm, so I dont want to work on this myself, but I am happy to assist you if you decide to try make the change yourself.

wchill, 2020-09-03:
I have a solution, it's not the most pretty but I think it's an acceptable tradeoff. Basically you add it as the content parameter in the message send API call. I tried adding it as part of the embed description but unfortunately that didn't work to actually mention a user.

[image]

I can clean this up and then send a PR when I have some time.

cameroncros, 2020-09-05:
So basically, you want to send it as a plain message alongside the embed? I think thats a suitable compromise. Especially in conjunction with #129

cameroncros, 2020-09-06:
Released in v2.57

PR #130 (2020-09-06): Add custom embed support
[image]

Closes #117 #129
closed 2020-09-06 by cameroncros