cameroncros/OctoPrint-DiscordRemote #118: [Feature Request] Finished Notification After Cool Down

Jmlannan, 2020-04-21:
Hello,
It would be nice to have the finished notification be sent once the bed temperature reaches a certain temperature.

Thanks for your work!

cameroncros, 2020-04-22:
Currently, OctoPrint does not generate a notification for that, so DiscordRemote would have to manage that itself.

If you want to add this feature, please do and I will happily merge it.

Jmlannan, 2020-04-22:
I believe I have a work around. You can add end gcode to wait until cool so octoprint does not think the print is finished until the printer has cooled.

Just add this to the end gcode in place of the normal bed shutoff code:

M190 S40; wait for bed to get to 40C
M140 S0; Turn off bed

cameroncros, 2020-04-23:
Even better! I will close this request then.

closed 2020-04-23 by cameroncros

sg000, 2020-09-15:
> M190 S40; wait for bed to get to 40C

> 
> 
> I believe I have a work around. You can add end gcode to wait until cool so octoprint does not think the print is finished until the printer has cooled.
> 
> Just add this to the end gcode in place of the normal bed shutoff code:
> 
> M190 S40; wait for bed to get to 40C
> M140 S0; Turn off bed

Just want to point out for anyone else looking here that you'll want to use M190 R40 for the cooldown.
S40 is for heating only.