cameroncros/OctoPrint-DiscordRemote #121: BUG?: Bot uploading every image

GregoryGHarding, 2020-04-29:
hey all
dont know if this is a feature or a bug. 

hoever it is quite annoying.. if anyone pastes a screenshot or posts a picture in my discord the bot automatically grabs it and uploads to octoprint. i would like to stop this activity, for obivious reasons. 

i am aware of permissions to block the ability for the user to upload, but it still responds with auth error etc.. i want no response.. 

so is this a bug? a feature? and how can i disable any action/response from DiscordRemote when ANYONE puts an image in discord? 

thanks! :)

cameroncros, 2020-04-30:
It's a bug. It could be restricted so that only gcode is uploaded, but I didn't do that as I felt it was easier to allow uploading all types of files instead. I don't personally think it is a good idea to have your octoprint bot in the same channel as anything else, just to reduce the risk of someone accidentally interacting with your printer and breaking it or starting a fire.

A quick fix would be to change the code to only upload .gcode files. This could be done fairly easily, but I will need some time to do it.

GregoryGHarding, 2020-05-03:
im not too worried about the users interacting with the bot, i have the permissions pretty tight, its only a handful of people there anyway, its just super annoying to have to log into octoprints ssh and clear uploaded images

cameroncros, 2020-05-07:
If you have the time/skills to make the change yourself, please create a pull request, I am busy with other things, so I cannot promise I will get to it any time soon.

PR #162 (2020-11-20): Download only gcode attachments
It will close #121
closed 2020-11-28 by cameroncros