cameroncros/OctoPrint-DiscordRemote #122: Please add Improved Instructions / Warnings

sg000, 2020-05-09:
First, I think the plugin is an awesome idea. I've never used/created a discord bot, so I was completely unaware of a few things. These should be in the up-front notes on the Plug-in. Unless I'm mistaken:
A) You have to do a whole bunch of stuff on a machine that's always running. The BOT runs off that machine, not off the discord servers in the cloud. I presume this would be the Pi, Unfortunately, I spent hours working on my laptop before figuring this out.
B) The instructions on setting up the files is lacking a LOT of detail, even the link to the "minecracft" bot lacks detail. I had to go to the discord bot instructions on their website and follow page by page.
C) Setting up a bot is tedious, errors are VERY dubious and lack guidance. Google is your only friend here.
D) A lack of example files from the plugin install make this 100% more difficult than it should.  for something this complex, it should have real files in the git to download at a minimum.

I'm trying to help the next poor soul who's getting in over their head before they waste half a day with nothing accomplished.

Best of luck with this plug-in, as I said, its and awesome thing. If I had a print farm, I'd put more work into using it.

cameroncros, 2020-05-10:
100% agreed, however, I do not have the time to write up more documentation. I would greatly appreciate contributions here.

dshokouhi, 2020-05-10:
It would be impossible to keep up with all the changes that discord can make to a page like that, the initially linked instructions are from 2016 so for sure we will have seen UX changes lol.  Maybe we should use instructions like this? https://discordpy.readthedocs.io/en/latest/discord.html

I personally didnt find much difficult following the link on the readme.

cameroncros, 2020-05-12:
@sg000 Are these docs slightly clearer now: 9e11c1b73df4464275ff87bf318816c38bbebeee

sg000, 2020-05-12:
> 
> 
> @sg000 Are these docs slightly clearer now: [9e11c1b](https://github.com/cameroncros/OctoPrint-DiscordRemote/commit/9e11c1b73df4464275ff87bf318816c38bbebeee)

Definitely a better link than before for setting up a bot.
I also think adding a note at the beginning of the plug-in would be helpful. So many people that might want this plug-in are not programmers or have servers up and running. So, unless I'm mistaken (and please tell me if I am), something like this would be helpful:
"Note, using this plug-in requires building and setting up your own discord bot. This task will require some programming ability (though some good copy-paste examples are included) and the ability to run the bot on a server or virtual environment anytime you want the printer notices to occur. This could be your Pi or a separate computer that runs all the time. Either way, you will have to set up this environment for this plug-in to interact with discord. Note, if you set it up on your Pi, if you reinstall OctoPi from an image, you will have to go through the setup of the bot program on the Pi all again."

cameroncros, 2020-05-13:
Okay, some of that sounds good, but I am a bit confused about the "the ability to run the bot on a server or virtual environment anytime you want the printer notices to occur. This could be your Pi or a separate computer that runs all the time. Either way, you will have to set up this environment for this plug-in to interact with discord." part. You don't need to set anything else up, the plugin itself is the bot, and it runs on the pi. All you need when you transition to a new Pi, is to copy the channel id and the bot token across.

sg000, 2020-05-13:
> 
> 
> Okay, some of that sounds good, but I am a bit confused about the "the ability to run the bot on a server or virtual environment anytime you want the printer notices to occur. This could be your Pi or a separate computer that runs all the time. Either way, you will have to set up this environment for this plug-in to interact with discord." part.
> You don't need to set anything else up, the plugin itself is the bot, and it runs on the pi. All you need when you transition to a new Pi, is to copy the channel id and the bot token across.
> I am also a bit confused where you had to do any programming, you should not have needed to do that.
> 
> New change: [b00b700](https://github.com/cameroncros/OctoPrint-DiscordRemote/commit/b00b70093be7274f15cd2b834f2175ea5033f32e)

Ha ha ha! You just proved my original point exactly. :)
Of course I do feel a bit dumb now for not realizing, but I'm not sure anywhere in the readme or doc for the plug-in does it tell you explicitly that the BOT programming is encoded into the plug-in and the plug-in is the programmed BOT! I thought the plug-in was sending commands to discord and the BOT had to be setup and programmed to respond to those commands on the discord! When I very first setup the BOT to the discord (before any programming) and entered the key to discord, my BOT wasn't online in discord even though I had the discordremote plugin on. I'll have to try again now that I understand the relationship.

cameroncros, 2020-05-14:
@sg000 Fair enough, i think I will try make that more clear then! Sorry you wasted time going down that tangent. If you still need assistance setting up your plugin, I am happy to help.

3f79c498699aecd4ee176a90cf45987b70f1e634

sg000, 2020-05-14:
I got it working!
Here's a few tips for anyone else who happens to come across this thread:
1) The Channel ID is the ID of the TEXT channel within the Discord Server that the bot is communicating with, not the Discord Server. If I had done THIS right in the first place I might not have gone down the rabbit hole I did before.
2) The original help link https://github.com/Chikachi/DiscordIntegration/wiki/How-to-get-a-token-and-channel-ID-for-Discord but STOP after "Get the channel ID of the Discord text channel" section. Everything else is not needed on that page.
3) I added the OAuth2 permissions as shown in the new link above (https://discordpy.readthedocs.io/en/latest/discord.html) in this thread, but then removed them and didn't seem to affect the bot function, so I guess they are not necessary.

Sorry for wasting your time, as most of it was me not understanding the distinct difference between text channel and discord server ID's.

cameroncros, 2020-05-15:
Awesome, I'm glad you are now working. Dont worry about wasting my time, you have probably saved others from diving down the same rabbit holes.

I have made one more commit, and I will leave the docs as is from now on: a6ade1a957d60f097465b119435c8e9e88d9e5cb

closed 2020-05-15 by cameroncros