cameroncros/OctoPrint-DiscordRemote #123: Premission denied

thostie2002, 2020-05-23:
Hello how can i fix this?
Yes i have read the “manual” but cant find it 

[image][image]

cameroncros, 2020-05-24:
@thostie2002 The user ID needs to be the numerical form, which is a number like: 165259232034275409. The easiest way to get this, is to join a voice channel, and then right click your name, and click "Copy ID"

thostie2002, 2020-05-24:
Great thanks man!!!  , now it works

thostie2002, 2020-05-24:
Can you also give me a full list of commands that i can use?

cameroncros, 2020-05-25:
/help :D

closed 2020-07-18 by cameroncros