cameroncros/OctoPrint-DiscordRemote #127: Appears not to be working in latest Octoprint update ( OctoPrint 1.4.1, Python 2.7.16, OctoPi 0.17.0)

JamieS01, 2020-08-05:
[image]

I apologise if this is not the right place to post this, however I am not sure how to troubleshoot this problem and I use this plugin quite heavily so I would love to get it working again!

cameroncros, 2020-08-06:
This is definitely the right place, but i am very confused about what is broken for you. I have updated my octoprint last night to 1.4.1, and its fine. Can you get me the logs from octoprint?

JamieS01, 2020-08-07:
I'm really sorry, this is totally my mistake! When I was changing the permissions on my private server, I accidentally revoked the bot access to read messages and it totally broke it. Fixed now.

closed 2020-08-07 by JamieS01