cameroncros/OctoPrint-DiscordRemote #129: Some sort of embed editor / way to make multi-line edits

Hexical1, 2020-08-16:
Hello! I really enjoy this plugin, but I think it would be amazing to be able to edit the actual embeds themselves. 

I did see an issue about this that is closed now, but as far as I can see I don't see any way in the UI to edit embeds - and I've dug around the config and don't have a clue where to start. 

Thanks for the amazing plugin though :)

cameroncros, 2020-08-17:
That sounds like an awesome idea, but unless someone wants to work on it, i don't think it'll happen any time soon. Short term, the print message could be changed to be json or yaml, and use that to generate the embeds.

I do not have the time to work on this unfortunately, so I will not.

I will leave this ticket open so if someone has the time they can work on it.

wchill, 2020-09-03:
I can take this when I have some time.

cameroncros, 2020-09-06:
Released in v2.57

cameroncros, 2020-09-14:
Reopening, so that a second attempt can be made at this.

closed 2021-10-17 by cameroncros