cameroncros/OctoPrint-DiscordRemote #134: Release v2.57.0 has broken the plugin for me. Does not appear in plugin manager and no sign of it at all in the web interface.

JamieS01, 2020-09-06:
Just updated the plugin via the popup that appeared in the web interface, only to see the discord icon removed from the top bar. Checked the bot itself in my server and it is listed as "offline". 

Went to the web interface's plugin manager and the plugin doesn't appear. Rebooted the pi, in hopes that would fix it, with no luck. Went to reinstall the plugin via the web interface plugin manager and it kept listing "requirement already satisfied" as if the plugin is still installed. Rebooted the pi again and no luck.

cp2004, 2020-09-06:
Python 2 vs. 3 error. Works on 3, not 2 due to invalid syntax. This line:
```python
def upload_file(path, author=None) -> Tuple[List[Tuple[str, Any]], List[Embed]]:
```

here:
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/ab405a204a7ba51a41356ca046666ffc68f2e09c/octoprint_discordremote/embedbuilder.py#L232

wchill, 2020-09-06:
Try installing the plugin via zip file from before the type annotations:

https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/eb59642cc68ef40feb871766933266bb3b7c20d8.zip

cameroncros, 2020-09-06:
Urgh, I will remove the type annotations then :(

cameroncros, 2020-09-09:
v2.58 should work on py27 again :(

closed 2020-09-12 by cameroncros