cameroncros/OctoPrint-DiscordRemote #138: Snapshots with Updates not showing

ccanevit, 2020-09-10:
With release 2.57.0, and continuing with the fix of 2.58.0, snapshots with updates are not occuring. Snapshot command does work, however periodic updates, and start of printing are not sending snapshots.
[image]
[image]
[image]

sg000, 2020-09-10:
I will concur with this issue. No snapshots with updates or conclusion with the latest release.

wchill, 2020-09-11:
Have also noticed this and I think this is my bad. Will need to send out a bugfix.

forouza1, 2020-09-12:
Same. Can concur no snapshots coming across but I see snapshots on /status commands.

TGCFabian, 2020-09-12:
Think alot of people currently have the same problem, i thought i went crazy, adjusted all settings, reinstalled plugin several times, reflashed octoprint, just to find this lmao

cameroncros, 2020-09-14:
@wchill I think i should have done a better review, there are a lot of breaking changes you accidentally made. I am gonna revert to v2.56.0, and start re-implementing the code in a cleaner way.

TGCFabian, 2020-09-14:
Eventho of the breaks/bugs, i appriciate you guys working on the plugin :)
Sadly i dont have much discord python experience, otherwise i would have helped out :)

cameroncros, 2020-09-14:
I've reverted the code from the past two releases, and it should be working now.

ccanevit, 2020-09-14:
Thanks for working on this guys. I'll do my part and try to report legitimate issues for you.

wchill, 2020-09-14:
Yeah @cameroncros I think I got confused by how your custom Discord implementation was doing some things.

ccanevit, 2020-09-14:
I can confirm, after I uninstalled, and re-installed with the 2.59.0 version, that snapshots are now working.

cameroncros, 2020-09-14:
@wchill yeah, it confuses me at the best of times as well. Really need to document it better.

closed 2020-09-14 by cameroncros