cameroncros/OctoPrint-DiscordRemote #139: Octoprint not being able to send timelapse

TGCFabian, 2020-09-12:
Trying to use the /gettimelapse command results in the following log error

```2020-09-12 15:03:48,366 - websocket - ERROR - error from callback <bound method Discord.on_message of <octoprint_discordremote.discord.Discord instance at 0x76061a30>>: [Errno 13] Permission denied: u'temp.zip'```

Running the default octoprint setup for a raspberry pi, running from raspberry pi 3A, latest revision of octoprint and plugin

|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 385253376
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model A Plus Rev 1.0
|      octopi_version: 0.17.0
|  python:
|    pip: 19.3.1
|    version: 2.7.16
|    virtualenv: /home/pi/oprint

ltmetalman, 2020-09-14:
It worked for me

Have you opted yourself as being able to post commands?

[image]

TGCFabian, 2020-09-14:
> It worked for me
> 
> Have you opted yourself as being able to post commands?
> 
> [image]

I have given my userID the * arg

While in chat, if i do the official file name, it sends nothing, when i make typo, it says "Failed to find filename" thingy
Also noticed mine is doing mpg files and not mp4, but thats something for a other time for me to change
[image]

TGCFabian, 2020-09-14:
[image]
My config is setup like this
NOTE!!! The MyUserID is repalced with my real userID, so its not like that :)

ltmetalman, 2020-09-14:
Does clicking on the hyperlink do anything?

TGCFabian, 2020-09-14:
> Does clicking on the hyperlink do anything?

If i click on the link(And change "your.host.name" to the ip) it does download it fine, thats why i think its a bug
Are you on the latest version of the plugin?(I am atleast)

ltmetalman, 2020-09-14:
Yes, I am on 2.58.0
I differ from you because I am on a pi3b and windows, but I wouldn't think that would matter.
I wish I could help you more, but I am a programming novice. Hopefully this helped the creator, since they are already working on the snapshot problem as well.

TGCFabian, 2020-09-14:
it is probs connected to that, 
Probs a slight error in the uploading part of the code

Personally i use a Pi3A and windows too, but there isnt a real differnece between pi3a and pi3b, so that is not the issue, its code backend

closed 2021-10-17 by cameroncros