cameroncros/OctoPrint-DiscordRemote #140: Discord not Reporting Print Status

omac03, 2020-10-01:
Hello, I can connect to my Ender 3 pro running a SKR Mini E3 with the TFT35 Touchscreen using discord and get snapshots but when I poll status it says it is not printing when it is. Discord never report that the printer has started a job or is printing. I can connect and send commands over Discord so I am confused on what the issue can be. 
[image]

cameroncros, 2020-10-18:
I dont know what the problem is, do you have some logs? My plugin appears to work correctly for me.

omac03, 2020-10-21:
Where do I pull the log from? I am not seeing one. In Octoprint terminal I get no status messages. I just updated the plugin but the discord still does not see it printing. What command is it looking for?

omac03, 2020-11-07:
It is still not reporting the status of my prints. DO I need a certain gcode in my slicer? where do i pull the log from? The terminal log? Really need to get this resolved!

cameroncros, 2020-11-07:
The logs are in the octoprints settings -> Logging -> octoprint_discordremote.log:
[image]

omac03, 2020-11-07:
Thank you! I have attached the log I have.
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5504945/plugin_discordremote.log)

omac03, 2020-11-17:
I resolved the issue. It seems that when printing through the TFT35 touch screen Octoprint does not report printing status. I went back to the internal SD card and it works perfectly now. Thanks!

closed 2020-11-17 by omac03