cameroncros/OctoPrint-DiscordRemote #141: Python3 Update Script fails to re-install Octoprint-DiscordRemote

sg000, 2020-10-02:
When running the Python3 update script (https://github.com/cp2004/Octoprint-Upgrade-To-Py3), it reinstalls all the plugins, but Octoprint-DiscordRemote cannot update because (I think) the link after installing Octoprint-DiscordRemote is still for the old version DiscordRemote plugin.

I think this is the case because in the plugin list, its listed as the old plugin and not the new plugin even after using the Install button from the "Get Plugins" window that is clearly for Octoprint-DiscordRemote.

cameroncros, 2020-10-03:
Did you install from the plugin manager or via the git url?

sg000, 2020-10-03:
> 
> 
> Did you install from the plugin manager or via the git url? Looks like if you installed via the git URL then it wont get picked up by the plugin manager, and wont show in the upgrade list.

From the plugin Manager.

cameroncros, 2020-10-03:
Try reinstalling from the plugin manager. I had the same issue, but after re-installing from the plugin manager it transitioned correctly. I couldnt see DiscordRemote in the list of installed plugins until i reinstalled.

sg000, 2020-10-03:
Yes, that's what I had to do. I'm just notifying that its not cleanly updating during the python 3 script thats already doing a reinstall of all the plugins. I had 2 fail. This one, and another that has been archived. 
I think its important because I bet the next major Octoprint update will force a reinstall like the script does or something similar. And if you didn't catch it in the trace, you might not realize the plugin isn't installed anymore.

cameroncros, 2020-10-03:
I dont think there is anything I can do about this. If anything, i would report it to the script repo instead. Its unfortunate, but I hope there wont be another transition like the 2-3 transition any time soon.

cp2004, 2020-10-07:
Just came across this one, (I do too much scrolling of issues) and since I wrote & maintain the script thought I'd answer the question.

It seems to be because the plugin repository listing has exported the ID as `DiscordRemote` and the internal ID of this plugin is `discordremote`. Obviously, the two do not match. I think this might be a bug in the plugin repository though, since it seems to be generating the `plugins.json` list using the name of the plugin file, rather than the internal parameter `id: ` (which is correct)

I will look into it, and see if it *is* a bug, since this is at least the 3rd/4th plugin to have this problem.

closed 2021-10-17 by cameroncros