cameroncros/OctoPrint-DiscordRemote #143: Discord command not responding

ShiroKrow, 2020-10-09:
Hello !
I used this plugin in the past, all worked well. But after an update it vanished from my server and aftre failing at reinstalling it, I gave up. 
Now that it can be reinstalled again, I wanted to use the bot and server that I had created initially, but it doesn't want to respond, either from octoprint or discord itself. I've redone a bot and a server, still doesn't work.
Do you have an idea for why it doesn't ?

Thanks in advance !

ShiroKrow, 2020-10-09:
Also I know they are necessary to help but I don't know how to find logs thus if you see this it would be great to tell me how.

cameroncros, 2020-10-10:
To get the logs, go into your settings:
[image]
The download button is next to the delete button on the right.

ShiroKrow, 2020-10-10:
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5358745/plugin_discordremote.log)

[plugin_discordremotelog.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5358747/plugin_discordremotelog.txt)

ShiroKrow, 2020-10-10:
Thank you ! Here they are.

cameroncros, 2020-10-10:
Check your channel ID: `2020-09-02 18:49:52,821 ERROR Incorrectly configured: Channel ID must be 18 chars long.`,
Unless you fixed that already?

ShiroKrow, 2020-10-10:
My channel ID is 18 chars actually, thus I think I fixed that, but it still doesn't work unfortunatly.

cameroncros, 2020-10-10:
Is your bot allowed on your server? You may have to authorise it.

```
2020-09-02 20:07:11,723 ERROR Failed to send message:
2020-09-02 20:07:11,726 ERROR 	Response: 400
2020-09-02 20:07:11,727 ERROR 	Response Content: {"message": "Unauthorized", "code": 40001}
2020-09-02 20:07:11,732 ERROR 	Response Headers: {'Expect-CT': 'max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"', 'Content-Length': '42', 'Via': '1.1 google', 'x-ratelimit-reset-after': '5', 'X-Content-Type-Options': 'nosniff', 'Set-Cookie': '__cfduid=d39790fff11aa6a9f4f2006df80431fba1599073631; expires=Fri, 02-Oct-20 19:07:11 GMT; path=/; domain=.discord.com; HttpOnly; SameSite=Lax; Secure', 'CF-Cache-Status': 'DYNAMIC', 'strict-transport-security': 'max-age=31536000; includeSubDomains; preload', 'x-ratelimit-remaining': '4', 'cf-request-id': '04f1d0552b0000edebcea74200000001', 'Server': 'cloudflare', 'x-envoy-upstream-service-time': '12', 'Connection': 'keep-alive', 'x-ratelimit-limit': '5', 'x-ratelimit-bucket': '80c17d2f203122d936070c88c8d10f33', 'Date': 'Wed, 02 Sep 2020 19:07:11 GMT', 'CF-RAY': '5cc983351ad2edeb-CDG', 'Content-Type': 'application/json', 'x-ratelimit-reset': '1599073637'}
2020-09-02 20:07:11,741 ERROR 	URL: https://discord.com/api/channels/[channel_id]/messages
2020-09-02 20:07:11,743 ERROR 	Headers: {u'Authorization': u'Bot [bot_token]', u'User-Agent': u'myBotThing (http://some.url, v0.1)'}
2020-09-02 20:07:11,745 ERROR 	Data: {u'payload_json': '{"embed": {"color": 15919147, "fields": [], "image": {"url": "attachment://snapshot.png"}, "timestamp": "2020-09-02T19:07:11.344472", "title": "Hello hello! If you see this message, it means that the settings are correct!"}}'}
2020-09-02 20:07:11,748 ERROR 	Files: [(u'attachment', ('snapshot.png', <_io.BytesIO object at 0x6bff59c0>))]
```

ShiroKrow, 2020-10-10:
He is allowed normally, I passed the Oauth2, and last time it worked it was enough. Is there more to do to make it have access to the channel ?
He is actually not online on discord despite the fact that he is on the server.

ShiroKrow, 2020-10-10:
Here is what it is right now.
[image]

cameroncros, 2020-10-10:
```
2020-10-09 08:02:35,684 ERROR 	Response: 404
2020-10-09 08:02:35,685 ERROR 	Response Content: {"message": "Unknown Channel", "code": 10003}
```
Are you sure you have the correct channel ID? Should be like: 6976XXXXXXX1280297 (X's are numbers)

ShiroKrow, 2020-10-10:
Ah ok, I'm a idiot, I confused the ID of the server and the one of the channel, too many ID. I apologize. Thank you for your patience, your time and your help !

cameroncros, 2020-10-10:
No worries, I assume that your issue is now solved?

ShiroKrow, 2020-10-10:
Yes it is, thank you !

closed 2020-10-10 by cameroncros