cameroncros/OctoPrint-DiscordRemote #145: Commands are case sensitive

GlitchTech, 2020-10-11:
Discord commands for this bot are case sensitive. At least if you have command restrictions on. I have not tested if the user has * on their permission table. So !Status does not work but !status does (with identifier of ! for command start)

Behavior of standard commands for other bots are that the base commands are case agnostic. Bringing this behavior in line with other bots seems logical.

Arguments, specifically file names, are likely fine to stay case sensitive.

cameroncros, 2020-10-11:
Can't say I agree 100% with the "Behavior of standard commands for other bots are that the base commands are case agnostic." argument, but sure, this is a reasonable change.

GregoryGHarding, 2020-10-14:
.toLowerCase() should handle this. no matter how you type the command, it will be converted to lower case first

Zinc-OS, 2020-11-09:
This issue should be closed

cameroncros, 2020-11-10:
It hasnt been implemented has it? I will leave it open in case someone wants to make the change

closed 2021-10-17 by cameroncros