cameroncros/OctoPrint-DiscordRemote #151: Problem Sending commands from Discord

Jonnywayway, 2020-10-15:
Hi I bet you are super busy, but I'm looking for some help getting the discord remote working. 

I can get it installed and get messages to the discord page, such as the test page or when I do /status through the octoprint user interface. What i am having problems with is the bot responding to commands in the channel. Any ideas on what is causing this? 
I have included the logs, I can see that it is getting the message maybe, but is responding with 500. 

[octoprint_logs.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5387271/octoprint_logs.txt)

ltmetalman, 2020-10-16:
Did you allow yourself to post commands like they show in the tutorial?

Jonnywayway, 2020-10-16:
I believe I did. I put a * in both the user and commands section

ltmetalman, 2020-10-16:
Well there's your problem. You don't put * in the user field. You put your discord generated code in the user field. Go back to the tutorial.

Jonnywayway, 2020-10-16:
Did that and it still doesn't work.

ltmetalman, 2020-10-16:
Ok cool, mine works fine, so you're definitely doing something wrong.

Jonnywayway, 2020-10-16:
Lol yeah that's why I made this post

cameroncros, 2020-10-17:
@Jonnywayway What is the command you are trying to send?Does your bot have permission to read from your channel?

Jonnywayway, 2020-10-28:
@cameroncros Thanks for the reply, 

I'm trying to send /status, and yes I believe the bot has permission to read and write. 

I can SSH just fine [https://pastebin.com/dwAcPxbW](url) I think I got the right list.

cameroncros, 2020-10-28:
> I can SSH just fine https://pastebin.com/dwAcPxbW <http://url> I think I

Thats the list of packages in the system, im after the python packages in
your octoprint virtual env.
Do:
```
source oprint/bin/activate
pip list
```

Jonnywayway, 2020-10-29:
`
Package                      Version
---------------------------- -----------
argh                         0.26.2
awesome-slugify              1.6.5
Babel                        2.4.0
backports-abc                0.5
backports.ssl-match-hostname 3.5.0.1
blinker                      1.4
cachelib                     0.1
certifi                      2017.4.17
chainmap                     1.0.3
chardet                      3.0.4
click                        7.1.1
emoji                        0.5.4
feedparser                   5.2.1
filetype                     1.0.6
Flask                        0.12.5
Flask-Assets                 0.12
Flask-Babel                  0.12.2
Flask-Login                  0.4.1
Flask-Principal              0.3.5
frozendict                   1.2
future                       0.18.2
futures                      3.3.0
humanfriendly                4.17
idna                         2.6
itsdangerous                 0.24
Jinja2                       2.8.1
Markdown                     3.1.1
MarkupSafe                   1.0
monotonic                    1.3
netaddr                      0.7.19
netifaces                    0.10.9
Octolapse                    0.3.4
OctoPrint                    1.4.0
OctoPrint-Autoselect         0.2.0
OctoPrint-DiscordRemote      2.60.0
OctoPrint-Estop              0.1.2
OctoPrint-FileManager        0.1.3
OctoPrint-MultiColors        1.0.16
OctoPrint-Pushbullet         0.1.9
OctoPrint-Stats              1.0.0
pathtools                    0.1.2
Pillow                       5.0.0
pip                          20.2.3
pkginfo                      1.5.0.1
Preheat                      0.2.0
psutil                       5.6.7
pushbullet.py                0.11.0
pyasn1                       0.2.3
pybonjour                    1.1.1
pylru                        1.2.0
pyserial                     3.4
python-dateutil              2.6.1
python-magic                 0.4.15
pytz                         2017.2
PyYAML                       3.10
regex                        2017.6.23
requests                     2.18.4
rsa                          4.0
sarge                        0.1.5.post0
scandir                      1.10.0
semantic-version             2.8.4
sentry-sdk                   0.13.2
setuptools                   44.1.1
singledispatch               3.4.0.3
six                          1.10.0
sockjs-tornado               1.0.3
speaklater                   1.3
STL-Viewer                   0.4.1
tornado                      4.5.3
TouchUI                      0.3.12
typing                       3.7.4.1
Unidecode                    0.4.20
urllib3                      1.22
watchdog                     0.9.0
webassets                    0.12.1
websocket-client             0.40.0
Werkzeug                     0.16.1
wrapt                        1.11.2
zipfile2                     0.0.12
`

Jonnywayway, 2020-10-29:
That was it, old packages, I went through and updated them and it started working.

closed 2020-10-29 by Jonnywayway