cameroncros/OctoPrint-DiscordRemote #155: After server left untouched overnight, when i woke up the Discord Bot was offline.

stelladraco27, 2020-10-23:
I setup the bot the previous night, and when I woke up after about 7 hours the bot was offline, the little controller icon was orange instead of green, and nothing I could find would reactivate it aside from a server restart. Is this a bug or did I miss a button somewhere?

I'm happy to provide more info if needed, 
Thank you

cameroncros, 2020-10-25:
It was probably a bug, can you get the discord remote logs from octoprint and upload them here, I can have a look and try find the reason.

stelladraco27, 2020-10-26:
Sure, attached are 3 logs, here are my comments and things I saw when looking through them to find the issue. 

[Discord bot error log 10-23.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5436589/Discord.bot.error.log.10-23.txt)

stelladraco27, 2020-10-26:
Sure, attached are 2 logs, here are my comments and things I saw when looking through them to find the issue, as well as a new issue today.

[Discord bot error log 10-23.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5436591/Discord.bot.error.log.10-23.txt)
This is the original issue, pardon the weird name. I had saved the log once it happened in One Note and cleared the logging in OctoPrint, so I had to re-save from OneNote to txt format. Ill add a detail that I forgot to mention: When this issue happened (Until print finished and I restarted my rpi 3b+) I was unable to connect to the web interface or ssh. I decided to change the network to 2.4 ghz ssid as I think 5ghz was dropping occasionally. I think that maybe the rpi wasn't trying to reconnect to wifi after it dropped temporarily and was visible again, therefore was not connected to wifi.

[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5436597/plugin_discordremote.log)
This is a new issue as of time of posting. I am currently printing a part and see the usual notification from the bot that a print has started. Unlike the last issue where the UI for OctoPrint was locked because of (I think) the raspberry pi not attempting to reconnect to wifi after it dropped, I am able to access OctoPrint perfectly fine. 

I went to check discord to use the /status command and noticed the bot was offline and would not respond to the command. The weird thing was that even though it was offline it is still sending regular interval status messages.

Thank you for your help, hope what I am typing makes sense, tired af currently.

stelladraco27, 2020-10-26:
Here is a screenshot of discord currently. I used the send command feature (controller icon in status bar) to send "/status". You can see that even though bot is offline a status was still sent.

[image]

cameroncros, 2020-10-30:
There is a lot of `Temporary failure in name resolution` errors in your logs, can you ssh into your pi, and try running `ping discord.com`?

ccanevit, 2020-11-03:
I'm also seeing a similar issue with my bot showing offline. Some update messages are still sent, orange icon... if I send messages from discord, I get no response back, but if I send the command from octopi I do get updates in discord.

cameroncros, 2020-11-04:
@ccanevit @stelladraco27 Unfortunately, neither of your logs is particularly useful, can you upload your earlier logs?

ccanevit, 2020-11-04:
Which log specifically are you looking for, the discordremote log or the octoprint?

ccanevit, 2020-11-04:
Adding my oldest ocotprint.log
[Uploading octoprint.log.2020-10-29.zip…]()

stelladraco27, 2020-11-05:
I pinged discord and it worked as I'd expect it to. The issue it is still happening as well, just yesterday, icon orange, bot offline, still sending updates. I see a pattern, it seems to happen after the server has been running for a while. I know a decent bit about coding, it seems that there may be a possible timeout or lack of method that tries to reconnect? Though I didn't code this so i'm not sure.

Thanks again for being active and trying to help us with this, its nice to know that I am not the only one with this issue as well

cameroncros, 2020-11-07:
@ccanevit Your link seems broken.
@stelladraco27 If there are errors, the bot does eventually time itself out so that you dont hammer the discord server and get your IP blocked. If either of you can get me the logs where the initial error happened, I can help debug from there. It definitely isnt a very good discord bot implementation :D

stelladraco27, 2020-11-24:
Sorry for the late response. I thought the logs I sent included the initial error? What would I be looking for to send you?

ccanevit, 2020-11-24:
I'm also not sure what additional logs to send, as I sent the log where the
errors began which should include before

cameroncros, 2020-11-28:
@stelladraco27 I am not really sure exactly what to look for, but it should be a bunch of completely normal logs, and then the first actual error. Delete all your logs, and then restart, and as soon as it hits the error, upload all the logs you can find. I can sift through them.

@ccanevit The link to the logs you attached is broken, try clicking it for yourself, it just links to this issue.

Grorco-Linux, 2020-12-08:
First day using the bot while at work, went down :( Don't know if this snippet from my log will help or not.

https://pastebin.com/ikNXD9Ci

timsmithlx, 2020-12-17:
No logs sorry but also having the same problem, bot will be online for “an unspecified” amount of time, then the icon goes orange.

if I send a /status from octopi discord receives it but not shows offline

and commands sent from discord are not responded to as the bot is offline.

cameroncros, 2020-12-17:
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/discord.py#L360
^ Looks like this might be the cause, I will create a new release later
tonight and push it out, hopefully that will fix things for you all.

timsmithlx, 2020-12-19:
did you manage to find the root of the issue in the end?? 👍

cameroncros, 2020-12-20:
Need to push out a release, give me a few hours

cameroncros, 2020-12-21:
v2.65 should make your lives better, but if not, please let me know and I will keep investigating.

eliteSchwein, 2020-12-24:
its now always offline and doesnt respond to commands

cameroncros, 2020-12-24:
@eliteSchwein can you confirm that your not token and permissions are setup correctly? Your error has something to do with lack of authorisation, which either means your bot token and channel IDs are incorrect, or your not does not have permission to talk to your server/channel. Please open a new discussion and I can help you there.

eliteSchwein, 2020-12-24:
[image]
the bot has admin permissions

cameroncros, 2020-12-24:
Okay, looks like you do have the same issue then. I'm very out of ideas. out of curiosity, what do you mean by 2fa enabled?

eliteSchwein, 2020-12-24:
i have on my discord account 2fa enabled

cameroncros, 2020-12-24:
Ah, okay, that's definitely unrelated.

stelladraco27, 2020-12-25:
Sorry for the lack of responses. After you pushed your update I have not been having my old issue, the bot has been online perfectly ever sense.

eliteSchwein, 2021-01-07:
i tried to reinstall the plugin etc and still bot is offline

cameroncros, 2021-01-07:
@eliteSchwein can you open a new issue, and get a recent set of logs?

eliteSchwein, 2021-01-08:
okay i dont know why, but after switching to lan instead of wifi it works

cameroncros, 2021-01-08:
Bizarre, I run on wifi, and a pretty flakey one at that (metal shed is a
great Faraday cage), and have had no issues. Glad your problem is fixed
though.

veexik, 2021-01-12:
Hi
The log from 2021-01-09 is the day the error occurs. the log from day, shows the bot is offline. i then went to discordremote in octoprint, and unchecked "enable presence" clicked save, and the checked "enabled presence" and saving again. and then the bot was online, without error in the log
[plugin_discordremote.log.2021-01-09.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5804860/plugin_discordremote.log.2021-01-09.txt)

[plugin_discordremote (1).txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5804858/plugin_discordremote.1.txt)

cameroncros, 2021-03-12:
@veexik Sorry for the silence, try release 2.67, it hopefully will either fix the problem, or give better debug logs.

closed 2021-10-17 by cameroncros