cameroncros/OctoPrint-DiscordRemote #161: Printing progress messages not sending

InterstellarRobo, 2020-11-03:
I don't know what's up but neither of the printing progress messages are sending. All the other messages will send and they do correctly. Any idea what's happening?

cameroncros, 2020-11-07:
Can you find the OctoPrint logs for Discord remote and upload them?

InterstellarRobo, 2020-11-07:
[plugin_discordremote.log.2020-11-02.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/5504960/plugin_discordremote.log.2020-11-02.log)

Here they are. On Nov 1st was the day I set it up so if you see errors from then that's because I had no bot setup but I have that fixed now.

cameroncros, 2020-11-28:
@InterstellarRobo Sorry for the delay.

At a quick skim, nothing looks out of place. What are you settings:
[image]

InterstellarRobo, 2020-12-01:
They are exactly that. I think I know my problem, my raspberry pi power supply is poor and is consistently under-volting. I am buying one soon and will update you once it arrives (which will be a while because I'm not paying extra for fast shipping).

InterstellarRobo, 2020-12-10:
[image]
These are my settings. I bought a new power supply and the pi is no longer under-volting but the images still aren't automatically sending. Is there anything weird you see here?

InterstellarRobo, 2020-12-21:
I updated and it started working again. Whatever you did, thank you.

closed 2020-12-21 by InterstellarRobo