cameroncros/OctoPrint-DiscordRemote #170: Feature Request: Send gcode files to octoprint over discord.

1iggy2, 2021-01-09:
I'm really enjoying the project so far. I'm wondering if you'd consider working on being able to send gcode to the printer via discord. I'm not sure if this is possible, but it seems like a useful feature.

Thanks for all you do!

1iggy2, 2021-01-09:
Also mentioned, but not replied to in #110.

1iggy2, 2021-01-09:
#66 and #67 are similar, but not the same as this request. I would be interested in the ability to upload gcode files to the printer over discord.

cameroncros, 2021-01-13:
Do you want to upload gcode files to the printers SD card?

1iggy2, 2021-01-13:
I would rather upload to octoprint. Give the user the ability to slice on a machine outside the LAN then send the gcode file over DiscordRemote to octoprint. Confirm it's available with /files and be able to start it with /print. All from outside the LAN without opening any ports.

cameroncros, 2021-01-13:
You can do that right now. Upload the gcode files to the discord channel
and it will get pulled into octoprint. You can also upload zip files and
split zips to work around the 8mb limit. You can use /unzip to unzip the
files.

1iggy2, 2021-01-13:
Oh wow, very impressive. I'm sorry that I did not see that. I will close this now.

closed 2021-01-13 by 1iggy2