cameroncros/OctoPrint-DiscordRemote #171: Bot goes offline every 30 hours or so.

groteboem, 2021-01-18:
**Describe the bug**
The bot goes offline every 30 hours or so. When I disable 'presence' and re-enable it again its online again. When its offline it still sends status updates but doesnt respond to commands.

[**Logs**](https://we.tl/t-YlFS6HPsHh)

cameroncros, 2021-03-12:
Sorry for the silence, try release 2.67, it hopefully will either fix the problem, or give better debug logs.

groteboem, 2021-03-14:
[Logs.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6136524/Logs.zip)

Its up to date. I dont know if this came with the new update or not, but now the bot doesn't want to go online anymore, even if I disable and re-enable presence again, or restart the pi. I attatched the logs. Thanks!

cameroncros, 2021-03-15:
Looks like the presence connection is not working correctly, unfortunately, I did not write that code, so I will have to investigate why that is broken. Probably an incorrect assumption somewhere.

closed 2021-10-17 by cameroncros