cameroncros/OctoPrint-DiscordRemote #172: Upload Files (Suggestion)

Dragonking3x, 2021-01-20:
I currently have two 3d Printers. One is at home and the other one at my Fathers place. I don't want OctoPrint to be open to the internet except with Discord. Sending him emails is connected with luck. If I don't call him, and tell him that I sent him a file, we will not find it in his email. Giving him a USB stick every time could be close to donating him a USB stick.

A Command like "-sendfile {filepath}" would be a nice solution to this. You would only have to send the command and include the file in the message.

I thought about searching other plugins for that, but if I have this one installed, it would be better for me if this plugin has that function.

cameroncros, 2021-01-21:
If you drop a file into the discord channel that octoprint is listening on,
it will be uploaded to octoprint. I think this is what you want?

Dragonking3x, 2021-01-21:
Exactly

cameroncros, 2021-01-21:
Well, I have good news, that's exactly what happens right now! But, this is
the second question about this feature, so it probably should be documented
better.

commit 5ac0fecf (2021-02-13): Fix #172 Documented file upload and unzipping
closed 2021-02-13 by cameroncros