cameroncros/OctoPrint-DiscordRemote #173: Wrong download Path in "-files" and  "-getfile" bug.

Dragonking3x, 2021-01-21:
The Download Path in the "-files" command is wrong.
If I use the "-getfile" command, it can't find a file or don't answer. I believe these 2 bugs are connected.

["-files" picture](http://drive.google.com/uc?export=view&id=1JPife60LOgvOJ2jslWfCN9IP50CT-N9d)

["-getfile" picture](http://drive.google.com/uc?export=view&id=1C2MjMfeWFlOMxDI6mTplj7WO7jHgw6R4)

It looks like, DiscordRemote doesn't like it if I make folders in folders.

cameroncros, 2021-01-21:
Can you try uploading and retrieving a file without underscores? I suspect
the underscores are messing it up. There may not be an easy fix on my end
for that though.

Dragonking3x, 2021-01-22:
It looks like that OctoPrint has the same problem now.
The underscore is not the problem. OctaPrind and DisccordRemote can handle it fine.
The main problem exists if I make a Folder in a Folder. It looks like, Octoprint looks at the folder, where the file is in, and only makes with the folder+FileName the download string. Because I have Folder+SubFolder+FileName, It doesn't work.

[Picture](http://drive.google.com/uc?export=view&id=1oWccwv6bYDlwPDjYGpwHNA9pE1JGioFd)

I will write OctoPrint about it, but if use the file name of OctoPrint, the bug should be fixed after Octoprint fixed the bug.

cameroncros, 2021-01-22:
Can you try doing -print with just the filename, and without the preceding
path? I have a feeling that may work, but I haven't looked at it yet. Let
me know if you find out anything.

Dragonking3x, 2021-01-22:
Sorry
I forgot to set the right's right for the last picture.

"-print" doesn't work if I only use the filename.

[-print](http://drive.google.com/uc?export=view&id=1Al9vR_tgsydJGPd-POMjRhWW1_GQt3hT)

The problem is this.
Normally the link would look like this.
192.168.1.57/downloads/files/local/**pat/test/wrench.gcode**

Sometimes, Otoprint had a problem and the link look like this.
192.168.1.57/downloads/files/local/**test/wrench.gcode**

The second link is wrong because it misses the "pat" folder in the link.
It doesn't happen all time, but I just found it after I tried to test the download with DiscordRemote.
If you use the file path that Octoprint gives you, you will have the same wrong link.
If you search for the file path yourself, it will be always wrong, if you only look at the foldername+filename.

btw. I use the Ocoprint image to install Octoprint on my Raspberry.

closed 2021-10-17 by cameroncros