cameroncros/OctoPrint-DiscordRemote #174: bot stays offline yet its connected

mmuziek, 2021-01-23:
**Describe the bug**
The bot is working from Pi > Discord but stays offline in discord so i cannot send discord > Pi 
Note it was working yesterday i have no clue i tried the disable and enable of the presense thing in the remote discord setting but nothing happens i do get the test messages in the discord chat channel

*Logs*
https://pastebin.com/RB4SwgvH

**To Reproduce**
Steps to reproduce the behavior:
1. start pi by adding power and turn on printer.
2. bot is offline and remains so

**Expected behavior**
bot comes online and listens to the commands

**Desktop (please complete the following information):**
 - OS: windows
 - Browser chrome

cameroncros, 2021-01-24:
This is a common issue for people, but unfortunately, I cannot reproduce
myself. I think the long term solution is to use discord.py instead of my
half-baked discord implementation, but i need to wait until python2 support
is no longer an issue.

mmuziek, 2021-01-24:
it might be a strange question but how about a button that restarts the bot?
Without affecting the print if its running? 
This allows us to refresh the connection for the bot?

cameroncros, 2021-01-24:
That might not be a bad idea, I will try find some time to do that.

closed 2021-10-17 by cameroncros