cameroncros/OctoPrint-DiscordRemote #181: @mentions don't seem to be working

Ocelot08, 2021-03-09:
Bot seems to write out "@ocelot08" but it does not work as a mention.

Also tried @ocelot09#XXXX and <@#[User ID Number]#> neither worked. 

I'm new to this so am I missing something?

cameroncros, 2021-03-10:
As per https://github.com/cameroncros/OctoPrint-DiscordRemote/issues/83,
mentions are broken inside embeds.

NarodGaming, 2021-04-22:
> As per #83, mentions are broken inside embeds.

What happened to the changes made on: https://github.com/cameroncros/OctoPrint-DiscordRemote/pull/130 (PR#130)?

I can't seem to find any reference to how to set this up, has this been removed since?

cameroncros, 2021-04-23:
I think it was removed, as it was broken. It didnt fix the mentions problem
though.

Prototype7813, 2021-06-14:
Hi ! Is it possible to find a fix to it ? I choose discord like main notification center and the mentions is the only thing that can notify me for real important stuff like the end of a print or some failure state. Would be really great since i have everything that i want with your plugin except that!

cameroncros, 2021-06-14:
I am working on using discord.py, which may help, but I can't promise
anything at this stage.

closed 2021-10-17 by cameroncros