cameroncros/OctoPrint-DiscordRemote #182: Bot always offline on Discord

pacodtc, 2021-03-14:
**Describe the bug**
After following the proper instructions and providing DiscordRemote (v2.67.0) plug-in with the requested Bot Token, Channel ID, and Base URL the bot remains offline and it is unable to receive or send commands from the discord channel. 

Performed the following troubleshooting steps unsuccessfully:
 - Regenerated bot token multiple times and updated the Bot ID each time
 - Created new text channels and updated the Channel ID each time
 - Attempted to use the hostname, the local IP address, and a blank field for the Base URL
 - Reinstalled plugged in and cleared data every time

The DiscordRemote command icon in Octopi always remained orange and the bot remained offline on discord server, unable to receive or send commands from the text channel.

*Logs*
Log of last attempt at reinstalling the plug-in below:
- Obtains an error about incorrect Channel ID, I verified the ID provided was accurate with the text channel I intended to use.
- A "failed to send message" constantly provided "Response Content: b'{"message": "Unauthorized", "code": 40001}'", I verified the bot had the proper permissions on my server and text channel.
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6136368/plugin_discordremote.log)

**To Reproduce**
Installed plug-in using Octopi plug-in manager. Followed plug in configuration instructions provided on DiscordRemote github README.md

**Expected behavior**
Green DiscordRemote command, Online bot status on discord server, able to receive and send commands from text channel.

**Screenshots**

- Verified the bot had proper authorization on discord server by running a sample python code to test read and write permissions.

[image]

- When using the sample python code shown above, the bot finally came online on the server and communication was successful. 

[image]

When using DiscordRemote plug-in the bot remains offline constantly 

- Octopi plug-in icon always orange

[image]

- Bot status on discord server always offline

[image]

cameroncros, 2021-03-15:
I think you have covered it, but can you just confirm that the Channel ID is an 18 digit number, something like: `467700763775205396`? You almost certainly have the bot token and permissions correct if the script you ran worked.

pacodtc, 2021-03-15:
Checked one more time, yes the ID is 18 digits long and corresponds to the same text channel I have used to run the script and show the screenshots above.

cameroncros, 2021-03-15:
Okay, interesting, I am having the same issue with my development bot i think, so I will keep working at it.

cameroncros, 2021-03-15:
Found the problem, websocket-client was bumped to v0.58, which broken this plugin. The new release of discordremote should now be live, and has a fix.

pacodtc, 2021-03-15:
Awesome! Thank you, the problem is solved. 

[image]
[image]

Although I did get and error about dependency conflicts with other plug-in while updating discordremote with plug-in manager. 

[image]

cameroncros, 2021-03-16:
Hmm, that conflict with OctoEverywhere is kinda out of my hands,
unfortunately websocket-client 0.58 is just completely broken for
DiscordRemote. In the future, I aim to replace my home-grown discord code
with discord.py, which may make things a lot better.