cameroncros/OctoPrint-DiscordRemote #183: Bot does not respond to commands from discord or send, when triggered from Octoprint

vfrdirk, 2021-03-16:
I have 3 Octoprint instances attached to Discord. Two of them work reliably, but this one is reluctant. I think it *used* to work right, but maybe a plugin was updated or something else is interfering.

Thanks for looking into this, and let me know what else I can provide!

[plugin_list.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6142673/plugin_list.txt)
[octoprint (2).log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6142676/octoprint.2.log)

cameroncros, 2021-03-16:
Can you try updating to v2.68, and confirm it is still broken? I do notice that your PI is throttling, which may be causing you issues? Might be worth trying a more consistent power supply.

vfrdirk, 2021-03-18:
I haven't forgotten about this, but my Octopi installation seems to have taken a digger. (Which may be related to the issue?) I'm working up a new install, so I'll see if this goes away with everything up to date.

twistedpixel, 2021-09-27:
I have a similar issue but the bot is responding fine when the command is sent from Octoprint UI, but ignoring commands I send through Discord (it worked fine a week ago - I added my user ID and * in the "access settings" section and it had been reliably responding for a few weeks whether I was on the local network or remote).

Any idea why it would suddenly start ignoring my commands?

cameroncros, 2021-09-27:
No idea, have you tried restarting Octoprint? I am still sporadically
working towards using discord.py, but I have limited free time, so progress
is slow.

twistedpixel, 2021-09-27:
Yeah I have restarted a few times since this issue appeared a few days ago but with no success. It's very strange but I suspect it's to do with permissions. I will persevere and if I make any progress I'll follow up here.

twistedpixel, 2021-09-27:
OK I have fixed it. For some reason the permissions settings on the text channel had reset and removed the role and bot. I added back the role and bot and now it seems to be working again. Possibly not the same issue as OP but resolved now anyway 👍

twistedpixel, 2021-10-03:
> OK I have fixed it. For some reason the permissions settings on the text channel had reset and removed the role and bot. I added back the role and bot and now it seems to be working again. Possibly not the same issue as OP but resolved now anyway 👍

Sadly it stopped working again without me changing anything. I'm not sure why this keeps happening.

cameroncros, 2021-10-03:
I have gotten the discord.py version of the server working, but it still
needs some polish. Once I have cleaned it up, I will be releasing Discord
remote 3.0 which should be far more reliable.

That said, having the settings reset is something I have not heard of
before. Is the SD card you are booting from okay?

twistedpixel, 2021-10-03:
> I have gotten the discord.py version of the server working, but it still needs some polish. Once I have cleaned it up, I will be releasing Discord remote 3.0 which should be far more reliable. That [...]

Oh cool, sounds good 👍

Yeah I believe the SD card is OK, everything else seems to be working as expected.

It just seems odd that I can send a "/status" command from the OctoPrint interface and the bot responds just fine but when I type the command in Discord chat, it is ignored. And this is 100% consistent. It's as if the bot stops listening to the text channel or loses permission to read the messages I type.

twistedpixel, 2021-10-03:
Oh I should have also said that the bot didn't seem to lose permissions this time it broke so I'm not sure that's actually what caused it the first time. It has correct permissions now and still doesn't work (although is sending me 10%, 20% status updates, etc. for the print I'm currently running).

cameroncros, 2021-10-12:
The bot can fail to receive messages without it impacting the ability to send. They are done over slightly different comms. Make sure your bot is allowed to read messages on the channel you want it to listen to (It might require more permissions, I am not sure. I just allow almost all perms). If it is allowed and still not seeing messages, its likely the broken plugin, which means you'll just have to wait a littlewhile for me to release the 3.0 version with discord.py integration.