cameroncros/OctoPrint-DiscordRemote #185: Roles Permissions

Bradly7412, 2021-03-29:
I am wondering if it is possible to use roles for permissions instead of just UesrIDs.

cameroncros, 2021-03-29:
Currently, no, it's user IDs only. but it wouldn't be too hard for someone to add that feature in.

Bradly7412, 2021-03-29:
were would we be able to find the file to add the Role permissions?

cameroncros, 2021-03-29:
https://www.github.com/cameroncros/OctoPrint-DiscordRemote/tree/master/octoprint_discordremote%2Fcommand.py

Look for the check_perms() function. It would likely require a bit of
manual testing to work out what info about a user you can get.

cOOM9009, 2021-03-29:
Hi im currently working with Bradly7412 to get roles to work instead
Is there a way we can contact you to help us troubleshoot any problems?

cameroncros, 2021-03-31:
I have a development discord here: https://discord.gg/znhGpkzp

Bradly7412, 2021-04-22:
@cameroncros Sorry for the huge delay but it would seem that the discord Invite that you have sent is invalid so am i able to receive another invite?

cameroncros, 2021-04-22:
https://discord.gg/bUTHJdBr <- expires after 7 days