cameroncros/OctoPrint-DiscordRemote #186: Octoprint Discord bot goes offline

DragonXC2, 2021-04-14:
For some reason, the Octoprint discord bot is offline and won't go back online. I've tried restarting the pi, remaking the bot, and updating the firmware, but nothing worked. Any suggestions?

DragonXC2, 2021-04-14:
The conditions are essentially exactly same as the ones discussed here: https://github.com/cameroncros/OctoPrint-DiscordRemote/issues/182 however my DiscordRemote is on version 2.68

cameroncros, 2021-04-14:
Check that your bot token hasnt been disabled or banned. You may need to
regenerate it.

DragonXC2, 2021-04-14:
I just regenerated it and it sent in discord that it was the correct settings

DragonXC2, 2021-04-14:
> Check that your bot token hasnt been disabled or banned. You may need to regenerate it.
> […](#)

The bot still doesn't go online

DragonXC2, 2021-04-15:
Are there no compatibility issues at all? Maybe WebSocket updated again?

cameroncros, 2021-04-16:
Websocket is pinned, so shouldn't have made a difference. Can you give me
the octoprint logs, and to be sure, the output of `pip list` in your
octoprint environment?

DragonXC2, 2021-04-16:
[image]

DragonXC2, 2021-04-16:
What is the output of pip list?

cameroncros, 2021-04-16:
Sorry, should have explained it more.

SSH into your octoprint instance, and then run the following commands:
```sh
source ~/oprint/bin/activate
pip list
```

closed 2021-10-17 by cameroncros