cameroncros/OctoPrint-DiscordRemote #189: Not sends blank messages

mB-PiBox, 2021-05-23:
The bot shows online. And when message is sent by the bot it’s completely blank in the channel

closed 2021-05-23 by mB-PiBox

cameroncros, 2021-05-23:
If you don't mind, could you explain what was wrong and how you fixed it? I
have never heard of that failure before.

mB-PiBox, 2021-05-23:
There was no issue on the DiscordRemote side; well not directly..

Apparently the bot sends the Information so that it requires "Link Preview" (show website preview info from links pasted into chat) to be Enabled. 

This is something that I personally do not prefer to be Enabled. As I've never been a fan of Meme/Gifs/etc,  in-which disabling that setting prevents Gifs sent through discord using the "GIF" button in the text box from being rendered. And flooding the channel. I prefer to see just the conversation of the channel instead of a bunch of images.

And since this setting can not be enabled/disabled per channel (that I know of), I've opted to just keep it disabled. And switched over to using Signal for print status updates.

Perhaps an option somewhere in DiscordRemote that allows the user to select sending of messages to the bot as plain text, instead of custom embeds.

mB-PiBox, 2021-05-23:
> If you don't mind, could you explain what was wrong and how you fixed it? I have never heard of that failure before.
> […](#)

There was no issue on the DiscordRemote side; well not directly..

Apparently the bot sends the Information so that it requires "Link Preview" (show website preview info from links pasted into chat) to be Enabled.

This is something that I personally do not prefer to be Enabled. As I've never been a fan of Meme/Gifs/etc, in-which disabling that setting prevents Gifs sent through discord using the "GIF" button in the text box from being rendered. And flooding the channel. I prefer to see just the conversation of the channel instead of a bunch of images.

And since this setting can not be enabled/disabled per channel (that I know of), I've opted to just keep it disabled. And switched over to using Signal for print status updates.

Perhaps an option somewhere in DiscordRemote that allows the user to select sending of messages to the bot as plain text, instead of custom embeds.

cameroncros, 2021-05-23:
Fair enough, I think you may find more luck with signal.
Discord doesn't make it easy to format the messages nicely without using
embeds, which is why I used them. It's possible to change the embeds to
playing text, but someone would have to spend the time to do that.

Thanks anyway though, is someone else has the same issue I know the cause
now.