cameroncros/OctoPrint-DiscordRemote #190: Only /status not working

Halzkomy, 2021-05-30:
I`ve been using Discord remote with my 3D printer for 2 days now. Everything worked fine until today. I shut down my printer and raspberry pi and then turned the power off. Later i started the printer and raspberry pi with Octoprint again. Everything seemed normal. My problem is that everything is working just fine except the command /status. Nothing happens when i type it in discord or try to use it via the Discord Remote symbol in Octoprint. I can pause and resume the print, /help is working and so on. The only thing that seems to not work is /status, which worked fine until today.

Log file:
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6566067/plugin_discordremote.log)

Screenshot of other commands:
[image]

 - OS: Windows 10 Pro
 - Browser: Chrome

cameroncros, 2021-05-31:
What was wrong? Did /status start working?

Halzkomy, 2021-06-01:
Yes, it started working about 1 hour into another print i started.

cameroncros, 2021-06-01:
hmm, that still sounds like a bug

stheyounger, 2021-06-25:
This just happened to me too. I finished setting up and only /status doesn't work. Same as Halzkomy's situation.

cameroncros, 2021-06-25:
Can you please provide any logs?

stheyounger, 2021-06-26:
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6719563/plugin_discordremote.log)

stheyounger, 2021-06-27:
/status started working again