cameroncros/OctoPrint-DiscordRemote #191: Plugin no longer working, won't reinstall.

Grauwulf75, 2021-06-06:
Just wondering if I'm missing something, but this plugin has stopped working and seems to have been automatically uninstalled from 2 octoprint installs that I run. I suspect this happened at the same time as the last Octoprint update. Last time I had it working was May 25.

It will not successfully reinstall, it looks like it goes through the motions of installing and after the reboot, it's not actually there.

cameroncros, 2021-06-07:
Can you give me any logs from the octoprint startup? And the kids from the
installation of the plug-in?

diebolo, 2021-06-07:
I'm experiencing the same issue, also after updating OctoPrint itself. Here's a text file with the relevant Traceback when starting and when trying to reinstall.
[octoprint_partial_log.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/6609443/octoprint_partial_log.txt)

cameroncros, 2021-06-07:
Ah, I see your problem now. Looks like the last change broke python 2.7
support accidentally.

Python 2.7 is no longer recommended by octoprint itself, so I suggest you
upgrade your octoprint installations via instructions here:
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

I will push one more release to fix things up for python 2.7, but I am
working on a new major release for DiscordRemote which will completely
break 2.7 support soon anyway, so I suggest you upgrade your octoprint
python version anyway.

closed 2021-10-17 by cameroncros