cameroncros/OctoPrint-DiscordRemote #194: Presence control is inverted

EvilGremlin, 2021-06-17:
2.69.0, nuff said

cameroncros, 2021-06-17:
Gonna need you to elaborate, I have no idea what you are talking about.

EvilGremlin, 2021-06-18:
I don't know if it's even possible to elaborate...
[image]
It's function is inverted.

cameroncros, 2021-06-18:
So, I think what your saying is that when the "Enable Presence" button is
enabled, the presence is not update, and when you disable it, it is?

I haven't managed to replicate it yet. From inspecting the code, it all
looks correct to me. This may be yet another quirk of the home grown
discord bot, which I am currently removing.

Once version 3 is released, I will ping you on this ticket to test, it
should work far better then.

EvilGremlin, 2021-06-19:
For me it's perfectly repeatable, I have this issue on 3 different machines.

cameroncros, 2021-06-19:
Okay, I will investigate more thoroughly then, thank you

cameroncros, 2021-06-20:
Yup, my printer also has it inverted, interesting.

EvilGremlin, 2021-06-21:
Okay, finally got some printing to do and when presence sorta works is always show "playing None", which mens it doesn't work at all...

sonjamichelle, 2021-08-22:
Sooo, Bot Presence is a no-go for a while?

I still get progress images in my discord so that is good. Commands don't, but honestly I never used them. I'm usually sitting at my desk next to my printers and have external access to OctoFarm and OctoPrint if I need to cancel a print or object.

Hopefully ya find time to work on the plug-in, it is a really neat plugin!

Thanks for your contribution and hard work!!!!

cameroncros, 2021-08-22:
@sonjamichelle Short term yes, presence may be broken. But, if you can't get commands working, it is likely that it is part of your problem to begin with. Check and see if there are any errors in the logs, try restarting and see if it responds to commands initially? If you have commands working, presence should also work, but the checkbox needs to be off instead of on.

sonjamichelle, 2021-08-22:
@cameroncros It looks like I had a typo in my base url. When my current print is done I'll fix and check my settings then do a reboot.

Also I don't seem to have allow_scripts: true in my config.yaml. Is that the config file? And where do I insert.

Odd, everything was working a while, but I DID make lots of changes to my OctoPrint install the past month or so. Soooo some things may have gotten buggered, lol.

closed 2021-10-17 by cameroncros