cameroncros/OctoPrint-DiscordRemote #197: Buttons!

DigiVore-Programing, 2021-08-09:
**Is your feature request related to a problem? Please describe.**
No, but it's something useful.

**Describe the solution you'd like**
Discord has added buttons to the UI, and discord.py has it now. Buttons would simply the bot

**Describe alternatives**
Reaction controls and such

cameroncros, 2021-08-09:
I still haven't completed discord.py support, but maybe once that is done it could be looked at. I am not particularly keen to work on this though, but I will accept any contributions.

DigiVore-Programing, 2021-08-09:
If you would like, me and a friend could help you switch to Discord.JS,
which is better in many ways, and works better. However you do need to
install Javascript I think...

cameroncros, 2021-08-09:
Discord.js is not gonna work, OctoPrint is python, so plugins need to be
python. Only reason I am using my own hand-rolled discord client instead of
discord.py is because of Octoprint not supporting python 3. If you have any
experience with discord.py, I'll happily accept PRs.

DigiVore-Programing, 2021-08-09:
Another idea, being able to move the extruder before every snapshot (like octolapse)

I'm gonna close this issue, since you've got the idea, and I might PR it.

closed 2021-08-09 by DigiVore-Programing

cameroncros, 2021-08-14:
Thats a pretty interesting idea, but may not be terribly easy, as far as I know, octolapse modifies the gcode before its sent, so that might be as easy for this? It might be possible though, but I will leave that to you.