cameroncros/OctoPrint-DiscordRemote #208: Print Scheduler Support

jneilliii, 2021-11-18:
**Is your feature request related to a problem? Please describe.**
@SgtKiLLx reached out to me about integrating print scheduler job list functionality for discord remote.

**Describe the solution you'd like**
A new command to list scheduled jobs.

**Describe alternatives you've considered**
n/a

**Additional context**
I was curious as to why there was a need for an API for this to work and attempted a non-API approach for doing this and was wondering if this would work or not? Since I'm not super familiar with the overlying structure of your plugin, I thought this might be a good place to start. 

Here's is where I'm at so far, but don't have a Discord server to test with completely, and SgtKiLLx was getting permission denied errors during initial testing.

https://github.com/cameroncros/OctoPrint-DiscordRemote/compare/master...jneilliii:feature/printscheduler?expand=1

cameroncros, 2021-11-18:
That looks pretty good, i will check it out and try it

SgtKiLLx, 2021-11-18:
> That looks pretty good, i will check it out and try it

I see it works. Will there be an update pushed soon?

cameroncros, 2021-11-18:
Just gotta get a small fix in, and once merged, should be released rapidly
afterwards.

SgtKiLLx, 2021-11-18:
@cameroncros , @jneilliii made that fix you mentioned. it works. its perfect. you can merge & push the new release.

cameroncros, 2021-11-18:
I'll try do it shortly, but if I run out of time I will have to do it after work. Thanks @jneilliii @SgtKiLLx

jneilliii, 2021-11-18:
hold off before merging @cameroncros, I'm going to give a stab at adding the ability to schedule jobs with a bot command as well.

cameroncros, 2021-11-18:
Works for me :D

jneilliii, 2021-11-18:
one quick question, how to the parameters work when sending a command? for example if I need to include a path (may contain spaces) and timestamp along with the /addjob command?

cameroncros, 2021-12-18:
Sorry for the delay, I have now merged this to dev, and it will be in the next release. Thanks for the contribution

closed 2021-12-18 by cameroncros