cameroncros/OctoPrint-DiscordRemote #212: Bot won't respond to Commands

rfreedman5280, 2021-12-16:
Hi there,
I've had a ton of success with this plugin for about a year now, but I'm having some issues with the bot now. It sends regularly scheduled status messages but won't recognize the /status or /help commands. I tried changing the command symbol to $ with no change. I also reset the bot token in discord and updated it, with no change. Any tips? I'm running a print right now so there's a chance restarting when I'm done will fix this.
Thanks!

cameroncros, 2021-12-18:
Yup, I have noticed this as well :( I will look into it today

rfreedman5280, 2021-12-18:
Thanks so much! I appreciate your help!

cameroncros, 2021-12-18:
I have pushed some fixes that should hopefully make the code a lot more reliable. I noticed that when you change the settings, the bot goes offline, and doesnt come back. It will now correctly come back. Hopefully that was the cause of both of our issues, and I will close this ticket. If the issues persist after updating, please re-open the ticket

commit 1cc9e8b5 (2021-12-18): Fix #212 Hopefully fix reliability issues
closed 2021-12-18 by cameroncros