cameroncros/OctoPrint-DiscordRemote #214: ERROR - Failed with: 'NoneType' object has no attribute 'send'

berkeoznalbant, 2022-01-26:
Discord bot seems online in the server but doesn't respond to any commands. It was working a while ago and stopped working all of a sudden. I haven't changed anything in the plugin settings but updated other plugins and also the octoprint software. I can't send command through octoprint either.

*Logs*
Please find the logs and upload them here. It'll likely be the first thing that is asked for, so save some time and upload them. Logs can be found under Settings -> Logging -> plugin_discordremote.log

I have attached the octoprint logs.

[octoprint.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/7944227/octoprint.log)

System Info:

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.99 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1400.0
env.hardware.ram: 917016576
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 21.3.1
env.python.version: 3.7.3
env.python.virtualenv: True
octoprint.safe_mode: False
octoprint.version: 1.7.3
systeminfo.generator: zipapi

cameroncros, 2022-01-30:
Sorry, didnt see this until now.

Do you know what plugins were updated? Try uninstalling DiscordRemote and reinstalling, and see if that helps?

berkeoznalbant, 2022-01-30:
OK that was my bad. Channel ID was changed without my knowledge. I thought I checked its ID but probably I copied the wrong ID. Now it works flawlessly. 
Thanks for the support though.

closed 2022-01-30 by berkeoznalbant