cameroncros/OctoPrint-DiscordRemote #218: Embed GCODE Thumbnail

Dealman, 2022-04-29:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
It's possible to have slicers include a small thumbnail of the model, stored in the GCODE file. It would be real neat if we could have the option to embed this thumbnail when a print is started - instead of showing a still of the printer.

**Describe alternatives you've considered**
None.

**Additional context**
This is enabled like this in Cura:
[image]

cameroncros, 2022-04-30:
If you want to add this feature, that would be great. I am happy to merge new features, but I wont development them myself.

Dealman, 2022-04-30:
> If you want to add this feature, that would be great. I am happy to merge new features, but I wont development them myself.

Uh I might have a look at it, I've never worked on a plugin for OctoPrint before so it would be quite the journey to learn how to go about testing to make sure it works before I make a PR.

That said, it should be a rather easy implementation so long as all the slicers follow the same concept.

cameroncros, 2022-04-30:
It isn't terribly difficult if you have python ability, I just don't have a lot of time to do it myself. I'll leave this ticket open in case you or someone else wants to work on it

Dealman, 2022-04-30:
I'm not a huge fan of Python but I know my way around it, if you could help me out with a few pointers I could probably have a go at implementing this later as I think it's a pretty neat feature. It may also be possible to do this via scripts, I guess? I set `allow_scripts` to true - but am unable to find documentation on how to work with this;

>Here you can specify a file to be executed before and after each message is sent to Discord. The file should be executable by the user under which OctoPrint is running

Would this be the place where one would implement a method, for example `get_thumbnail`? If yes, how would I go about fetching the GCODE to parse it when a print is started?

https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/de7962a0dddec7928a281c451de971b5b7d3ce97/octoprint_discordremote/__init__.py#L530

cameroncros, 2022-10-11:
For what it's worth, I didn't know/use python before I took on this project
either in the first place.

I don't really know what the scripts stuff is for to be honest, it was
already part of the old plugin, and I don't know if it still works or not.

I would start looking around here:
https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/__init__.py#L506

You probably want to add a special case for "printing_started" (see
`on_event` function for all events), and add a `get_thumbnail` function
that gets called there. I dunno how to get to the gcode file and pull out
the thumbnail, you may have to work that out yourself.
There is octoprint plugin development doco somewhere that will help you
though.

https://docs.octoprint.org/en/master/plugins/index.html

Dealman, 2022-11-27:
[image]

Made some progress! Some clean up to do, and need to investigate other formats as it only currently works with Cura. I was gonna make it inside a separate python script, but I can't seem to import it whenever I try - do they *have* to be classes?

cameroncros, 2022-11-27:
Imports in python are weird, it can take some fiddling to get it right.

I would look at how the other scripts are imported and see what they are
doing.

a1co, 2023-05-21:
> [image]
> 
> Made some progress! Some clean up to do, and need to investigate other formats as it only currently works with Cura. I was gonna make it inside a separate python script, but I can't seem to import it whenever I try - do they _have_ to be classes?

Would love to see that feature merged. Is it going to come?

Dealman, 2023-05-21:
I think I have the code laying around somewhere for it, but you'd be compiling your own version of the plugin for OctoPrint. I only ever so slightly dug into making a proper PR for it but was quickly reminded of how much I despise Python so I started working on other projects instead. 😅 

If I recall correctly it, if it found a thumbnail in the GCODE it used that instead of a snapshop - so it wasn't a fully implemented feature but more of a workaround.

I can't make any promises, especially not since my printer is currently out of commission and as such finding motivation to work on this feature is quite low. 😞