cameroncros/OctoPrint-DiscordRemote #219: API Returns 403 Forbidden Error

Dealman, 2022-04-29:
**Describe the bug**
The API is unuseable because it requires some kind of authorization, which the documentation does not provide. Trying to use the API via Postman results in `HTTP 403 Forbidden` with the response `"Insufficient rights"`.

**To Reproduce**
1. Use Postman or equivalent, setup a POST following the documentation
2. Try and send the request, receive HTTP 403 error

**Expected behavior**
Expected to receive either the message or a sent command in Discord.

**Screenshots**
[image]

cameroncros, 2022-04-30:
Huh, to be honest, I have no idea about that part of the code, it worked when I wrote it, but that was a long time ago.

https://docs.octoprint.org/en/master/api/general.html

I suspect that doco is relavent to setting up the authentication, can you try it and confirm?

Dealman, 2022-04-30:
Had a quick gander at it, it now returns the message `Failed to send message` with a `404 NOT FOUND` response code.

Dealman, 2022-04-30:
[image]

Was able to use the API to provide thumbnails, however it's limited to rather small sizes due to the 6000 character limit. 128x128 is a pretty nice size, but results in ~6300 characters. Attached is a size of 92x92.

Any chance this limit could be increased?

cameroncros, 2022-04-30:
The character limit applies to strings, you can instead attach the images
the same way snapshots are attached, and then it shouldn't matter about the
6000 char limit? I'm not 100% sure how your sending the images to be honest.

cameroncros, 2022-04-30:
Oh, sorry, your talking about the Web api  and sending messages that way?
That might require some changes in that case. How are you sending the
images to the api? Can you show me the data you are sending?

Dealman, 2022-04-30:
This is how I sent the above image;

```json
{
    "command": "sendMessage",
    "title": "Title",
    "description": "Message",
    "image": "iVBORw0KGgoAAAANSUhEUgAAAFwAAABcCAYAAADj79JYAAAACXBIWXMAAA7EAAAOxAGVKw4bAAAMFklEQVR4nO2be3BcVR3Hv79z7+7m0TZN05A+UhuSUKCEbHgNVRHrOMprKK0z1SkiooOvcdSiIx2fHf/wNYAIg4PAMA4M8qgIWCyICNRuKlZbiyFgk+bRPDbZPJqkbbKPe+49P/+4u9nNZvPebWDmfDIn2ce95/x+3/M7v3PuuTeARqPRaDQajUaj0Wg0Go1Go9FoNBqNRqPRaDQajUajed9Di23AQnlkd3X50sKC7xO4DgSwonclcNfndjU0AcBT99Ssc2zBt+xq6F5sW4H3ueDf+0Z1ae3a/KMlRcbaMxEFpYDCPMJYlIfOhNV/833kZ2AFAWMMuvPmXQ0PAuDFtPl9KfhjP7ugxEPC3zPI12ys9N559WX5KCj1AgRYQxKHG6N4q9lCSZEx4TwivMaOdfuO7x07sTiWv8cF370bYkPhhVUe4fE7iv2GoDpm9gMoBxMNn3EAAFfW+LCqxAQRYWDYxtGmGAZHFNaUmpPqJPAZMN/5mV2Nvz3L7gAAJlu0SOzdvbogWlRcA2XUgakOQC2BaxVoKTNgEAEMUDxGOB4qHP/FDDAzmAFHud9RxnCipSB68Jm7areqmPrSjh82duXeuyRZFfzJn9aUefJwBQHLmURH46mGN3/yE9jpxz31y/PXmIbXD0YtE11CYH8MVA0Vt2dcTJpyCBIASzI8JgFE4HhmZgYUAx5zKsHHucbME2/vuavmO5/+buOj8/V5rmQtpTy8e+O3VxaZPwfBm/jMsvBWz6C1s6LMU65Afib2A+QHo2zBVjIQOmnDaxKurMlDabGbrwdHHBz+XxSRGKNsxeziiZlfcpT1pR27mnrmZNc8yEqEf//L1ZdWrPbes67MRJQBMgSWmAwrxnUxqfareKiNx2s2upkAQ7h/U5cdKh7qzDNGeLIqouuF8DU+d0/tzk99p+HxLFg3JfMRnP5494UfIGHWMXMdiPy9g86lA8MOaqq9WF/mBZkC4SEL7zbHcKJXoqbKl3XDAcDriaeS8XTCAAO2A3hMmrXgcYoZeOzZe2q3ewx1+007G/uyb/EMsfbS/dW+iFNwETH7AaoD2E+gWgaKU487edrBWISxqSYPq0oMeAzC6bDC8U6JhtYYNp7rnaqJhRHP16tKDJyz3ASDMTjiIDjgwDRm8i5ZRzpEGATwzW13NDyVZYuTEb7nZ9WlHl9BLQtcQqBaMNdFHbqAAE8yVAiMyX54TIK0FZRyVwkJH4QApJxu6lsgBBgEDAwrDAxbE+yZSx0ZWAngyed+7d8eHXO+dvMPshft5iO7L3qhdLl5OYjXMFNyFFKaTNP4YAqC47jRllgtELnnmMach/aiQPFfPDHitxUUGB9+4d7ar2+9o+HZbLQjDjZEN41G1FqAiNwVVuaCqYtpAk5adBPc6JM2T13ne6CcCSsYgmEY7iRsGoBhAFFLJTrhHBD94U/3+Z/Ze/eGlQsWPBxW39z/n7ATtdS8jc7zEmIWuxccCdHjvWEYNG1n5VzUadoGGB4TOG+dF+VrTKxYaWJliYF155goyBOIyaQmAD7NZt47e++7+KYFCb7nlfY9Z8bUzv1HIixtnpcQggDTpPiVHsevCAFBhII81+v5CJKVMkW7iLfd0i3h8xIuPD8fV25egcuvWIJ1q0xEYwpj4bTRKegcInr+xfv8T+z51cYV8xHcAIB3Wkb+VVlRZJwcUR9dv9qEIWaIykmOEFYWGTAFwfQQHAWMRhTGIgrFywQog7JnO3Vkat+S7sXT0kIBjimEB2MYDEm0dttobLWwrFCgIE9MCAyACIRarzBu3XFdWfNTL/c1z0VwkXixZ1/7j/uH7YcPNkTAmEXeTfPB6yFYNiPYb6MtKBE66cBRbpTPRoBsdcRc6kV8zUUpgjDc9bytGKZJUwQYAMJqIvzpz/f7f/f8vRXL5yw4ADy9r/1rwX7n+UONUSTyQq7zKKX9zHT03GqbvkZpT9xz4fgqy1GAbQM+T+ZgSRnZBIHbvJ6it1984OLr5yw4ADWixI62Hrn/rWYrMYSSjlAuCkMId2VgGnBXDAKg2YyyWZWp2wbcY0SqCgywAiyb4TUznJfWjfHX5QaJP7/0gP+hl+6vXjYXwfHyyy2xyChve6fd+u+77dbEoTptbKWUWYrBzBg6pWAId+J1nScIAQydVnCc3C4ppe1O9ilag+HuxyiHYZoT/UoVYHJ6IgLRl2EsaXjlgbpPzFpwAHhh/4mRmHSuO9pktbUGZXrumlTmlkKSZTSsULHGxMZKLyorvVi3YQmqqny4uNqL8z7gQSTGs65rPiXR0WI8pyS3d5EYBdP5mbmsV4L/su83/t++8nht4awEB4DnXunoBfjaQ43R/q4+OW0j8zAKRO5Fx7ETFpQCVq/14dwLl2HVGi+kA7zdYmHwlJPTCC8pEihPuyuk4mFetdYz/fnTdiYJg+grGKWGlx+s25xaP2EGbr6h6jLTxBsfvyJ/6aqSWWwu0rRvJ9A3ZCM06OCS831YvdKEz0uwpEL/sMLRphh8XkJ1uWfmNudJ6lWxxwQMQXAchnQmXeIvoH5WYDxoGf13bvlKb3jKCE/w5L7WI7bNn9r/n0h06PQsIg4Ty0xIJ8Wz8b0YN3eLOaSnTGUmUo+TNhC1XLGB+Y2Y9MYT0U5EX8/jsrdefch/1YyCA8CT+9r+FovhC6/9O+ycCc9/C2DyRJM6UaXslKaIvhDFc5mOMnXshObT/WScpxTtn5XgruitT0cttfPVQ2FEYmrBKxQid52begN4XGuauGw7a2UO/Tllx6bETGICZgaUYoyGVfec7vg8sbf9gc9uqSp79VDkh9d9sAA+L818EhLWZfqYYDsp73nikUIk1rm5Ii1RT9FUpnROKV8kxB3/iN3d0+5+G82dEh0huzPUb9/d0iMfnfMttt/vbf3RLTdWrXrtcOT2azYVwDCSVs5VGkHJRxs4PaVwcsTkjsyVpws8flTKKEx9SgBwt6HbghLNnRLtPTLU0i0Do2NczyQPBIMDDQAUMM+byLEXW7/ad1NVyRtHIts+fkV+ch2bwcrp9CpeJrBmpXtrbNyn+N+iJSL+cM98LJwdPOmFC6UdMR4MKUERjiq0dLkCn+iRrce7ZSBqoZ7ICgSDg1NuaM3bne0fLM/3lXr/Ul3uvfqjl+QtMBQZPg/BNN07RzEZ3+bNaTqZwpJxcVPmFAZGRh00d0g0d0nV3iPfOdFjB6SDgIjFAt0nTwZnW/+CPNq6uWL50mXi7xdV+Wo3XeTL9fjPLtPkXwajf8hBU4eF451StvXKI90hJ6DgBMRo5GD36dND8212wQrdcs361SLP+MdlF+RV1G2Y4XGIxeyPlHSQnn+VYnT12WjqlGjtssZauuU/B4ZVPZMTUEq82dvbG86WGVmR4NbrKzfAS/VX1eaVnl/hXVRdU0nNu6lXjokJrqlDoiVoDbV2yvqRMaeeiAPBYN9hYPLjedkia9p8bkvl5Yag1z92ef7Sc9dOfTmeq85Ij9rUaI7E3Gdkmjokt/VYPce75YFIVNU7Dgf6+voaMWnazB1Z9f/zW8/9pBDixWs3FXgzPSqcCzKtIE6NOmjqkGjqsLg9aB8/3i0DilW9lOpAf39/21kxbAqyHnCf31q5w+OhJ274UKEoLc6W6JxMDcm37jecmOAkmjotpy1oN3T2ygAT11uWExgYGAhlyYiskJMRfuvWym8V+MS9N36kkJYvSfkvhNm2xjz9BNdvxwWWsZYuebhvyA4IxYGxWOzg8PDwqWz5kQtyNr/dtq3yF4X5YtfWq5egMH/6LZvU5RiQeYI71iFxvEOeaQnKN0dGnYDjcCAUCh0CEM2VD7kgpwuK27ZVPlq81PjijR8pRJ4vLnrqbux4akimjMQEd6xDoqVLDrR0W/XhGAeUQqC3t/coAGdSQ+8jcir45s0w1xdX/XHVCmPLlqsL3eZ44pLAneAsHDshuaVbdrUG7QO2VPUOcyAUCv0PZ3EFcTbI+ZJ5+/by/CW2769fuHHZVUIQ+odsHOuQONZhqZYu2dQVkgdYod4BDvT29nbm2p7F5qxco9y2tWJ5abH3vtePRC4InbQPKoUAM9eHQqGBs9G+RqPRaDQajUaj0Wg0Go1Go9FoNBqNRqPRaDQajUaj0WSV/wPxTk/z+BaDMwAAAABJRU5ErkJggg==",
    "imagename": "Thumbnail.png"
}
```
Also, the color parameter I wasn't able to make use of at all. Are you sure it's meant to be in the format of `0x######`?

cameroncros, 2022-04-30:
Does it fail if you send a larger image? The 6000 char thing only applies
to text (it's a discord limit, not mine).

Not sure about the colour thing, did you try sending the hex as a number as
well as a string? One of them should have worked.

"color": 0xAABBCC,

Vs

"Color": "0xAABBCC",

Or maybe:

"Color": "AABBCC", ?

Dealman, 2022-04-30:
Ah, I stand corrected. It does work with larger files, I guess when I tried last time it failed to reach for one reason or another. Tried just now with a 256x256 thumbnail and it worked just fine and dandy.

As for the color, the parameter name needs to be `Color` and value a string. However I don't see any changes in color anywhere in the message.

[image]

cameroncros, 2022-06-08:
Sorry, I have been busy, so not really following this much, but the colour field if it was working would affect that yellow strip on the left of the embeds in the screenshot you supplied. afaik, it should work, but I haven't looked at it deeply yet

Dealman, 2022-11-26:
Might as well update this as I've started getting back into 3D printing, the correct format is just a straight-up integer. As seen listed [here](https://gist.github.com/thomasbnt/b6f455e2c7d743b796917fa3c205f812).

The plugin itself uses hex in the format of `0x00AE86`, this format is incompatible when using as the API as is `#00AE86`. The API expects an integer, not a string. In order for this to be compatible with the API, the format needs to be `"color" = 44678`.

IMO this is very counter intuitive, and seems like an oversight/mistake?

I might look into this after I'm done trying to implement #218.

cameroncros, 2022-11-26:
Ah, thanks for finding that. It might be a simple fix in that case, use `int()` as described by https://stackoverflow.com/a/209550

Dealman, 2022-11-26:
> Ah, thanks for finding that. It might be a simple fix in that case, use `int()` as described by https://stackoverflow.com/a/209550

I'll check it out for sure, I'm currently trying to figure out how to build this. First time touching plugin development for OctoPrint, so I'm currently just uploading a .zip file over and over again - it refuses to accept me linking to my forked repo.

Any suggestions on how to set up the build environment properly? 😅

cameroncros, 2022-11-26:
If you've forked the repo, there is the test.sh script in the top of the directory that you can use to spin up a virtual environment with octoprint and the plugin installed. itll also spawn a webbrowser pointed at octoprint.

Is that suitable for testing?

Dealman, 2022-11-26:
Thank you, that should help 👍