cameroncros/OctoPrint-DiscordRemote #220: Bot Offline?

Dhalkar, 2022-05-08:
So the bot worked earlier today.
Now it's just listed as Offline, it works fine to invite it, I've also tried reinviting it and redoing the entire process.

Is the Bot itself offline?

cameroncros, 2022-05-09:
The bot is your octoprint server. Can you get me the latest logs from your
server so I can have a look?

Dhalkar, 2022-05-09:
> The bot is your octoprint server. Can you get me the latest logs from your server so I can have a look?
> […](#)

I checked the logfile myself, and found the error, for some reason my Bot Token is not 59 chars, it's 70.
The only thing I've actually changed is the bots name and I've added a picture instead of the default one.

cameroncros, 2022-05-09:
Try regenerating the token and re-setting it in octoprint.
If the length is still wrong, can you send me the bot token? (Reset it and
send me the old one)

Dhalkar, 2022-05-10:
> Try regenerating the token and re-setting it in octoprint. If the length is still wrong, can you send me the bot token? (Reset it and send me the old one)
> […](#)

Tried generating a new one and setting it in OctoPrint, still get the longer tokens when generating a new token though.

EvilGremlin, 2022-05-18:
i confirm that discord stably genetrate longer tokens now (and i have 2FA which made testing just annoying)

KenwoodFox, 2022-05-19:
Just in, im also generating 70 char tokens, will it also generate 50 char ones or is a change in the code needed?

cameroncros, 2022-05-19:
It's a trivial code change, but I will need to find some time to do it.
Short term, if you can find your plugin source code on your octoprint
installation, look for this file:

https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/discordimpl.py#L85

And delete that block (the if plus the the following 2 lines).

cameroncros, 2022-05-22:
See latest release

closed 2022-05-22 by cameroncros

cameroncros, 2022-10-11:
Aww, nice of them to change that for no good reason. I will try push out a
fix later this week

commit dc7e3376 (2022-05-22): Fix #220 Remove token length validation