cameroncros/OctoPrint-DiscordRemote #224: Printer or Profile ID

WWE-Corey, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
I run 3 instances of Octoprint with DiscordRemote tied to a single Discord Bot. It is not clear which printer is posting activity.

**Describe the solution you'd like**
Include a method of naming the printer or use the Profile Name when sending updates to Discord.

**Describe alternatives you've considered**
I see the IP/Hostname option but it is not sending any data (blank) and those are not very clear anyway.

**Additional context**
n/a

cameroncros, 2022-08-20:
There are a few options that may suit.

You could create another 3 discord bots, and then name them as you please.
You could also set the title for the octoprint server, and it should be used as the author field, which goes at the top of the embed:
[image]
[image]

The IP/hostname stuff is exclusively for the /status command.

WWE-Corey, 2022-08-20:
Thank you - I assumed as much for the other options and went the route of 3 individual Bots. Thank you.

cameroncros, 2022-08-20:
Cool. Sounds like you are all sorted then, I'll close this ticket

closed 2022-08-20 by cameroncros