cameroncros/OctoPrint-DiscordRemote #226: Discord bot will not respond to commands sent on discord

Jaddy4567, 2022-09-04:
For some reason I cannot get this to work. I can't find anything to help me in the logs but basically I can send a command from the Octoprint Local browser server but not from the discord server its self. I dont even get a permission denied response. It will not register a command from discord but it will from the raspberry pi running OctoPrint. I have attached the logs; yes this is the full log and all I get from hours of testing. I have also attached a screenshot of what is happening on discord. That first status command was sent from the raspberry pi. The Ip is hidden by settings not an error. 
Can anyone help me? Im very new to OctoPrint so I dont know what I have done. Thanks

[image]

[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/9484365/plugin_discordremote.log)

cameroncros, 2022-09-04:
This is most likely permissions. 

[image]

The default is to not allow any commands, make sure you have an astrerix in both fields. for the first row.

Jaddy4567, 2022-09-04:
Yes I have that, surely its not a permissions issue because im not getting any response at all. I have found other forum posts but I have tried all the "fixes" (that I could find) none worked for me. Any other ideas?
Its sort of the same as Issue #55 but again not a permission issue I dont think. its almost like it can't see my messages. I have checked and double checked the permissions the bot has on the server and for all I know it has all the permissions it needs. Unless im missing something. I think I may have just found my problem. Discord thinks there is no commands. do I have to configure this somewhere? (also are my permissions right?)
Thanks.

[image]

[image]

Jaddy4567, 2022-09-06:
@cameroncros. So I have found my issue. The discord application has no commands. what do you think I have missed when setting this up? I also have attached the OctoPrint.log file. I dont know how to read these sorts of Log files but I can see alot of "errors". I dont know how to fix this. Like I said im very new to this. 

[octoprint.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/9492714/octoprint.log)

cameroncros, 2022-09-06:
This plugin doesn't use discords command API, that was added a while after it was implemented. The plugin just listens for messages and parses them itself.

If someone wanted to move to discords command API that would probably be a good change to make.

cameroncros, 2022-09-06:
```
Error while executing SimpleApiPlugin discordremote
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 154, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/__init__.py", line 346, in on_api_command
    return self.execute_command(data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/__init__.py", line 356, in execute_command
    messages = self.command.parse_command(args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_discordremote/command.py", line 66, in parse_command
    parts = re.split(r'\s+', string)
  File "/home/pi/oprint/lib/python3.7/re.py", line 213, in split
    return _compile(pattern, flags).split(string, maxsplit)
TypeError: expected string or bytes-like object
```

This looks like it's probably your issue

cameroncros, 2022-09-11:
@Jaddy4567 I noticed that the discord.py plugin has been updated, and it seems to have been a breaking change. I have fixed discordremote to work with it, could you update and see if it fixes your problem?

cameroncros, 2022-09-11:
Are you able to get the output of `pip list` from your octoprint install?

cameroncros, 2022-09-13:
@Jaddy4567 

I think I have found the root of the problem. Discord has changed their permissions:

https://discord.com/developers/

[image]

You can find that under the bot settings. Enable all 3 and your bot should just start working again.

Jaddy4567, 2022-09-13:
Hey Cameron, I will get the Pip list and test that solution tonight when I get home. Sorry for not replying in a while. I almost gave up on getting octoprint to work. I will get back to you soon. Thanks

Abinash3366, 2022-09-14:
[image]
Just the heads up, bots require a different scope to use a slash command, This is probably the issue even if you give the bot administrator, it still requires the applications. Commands scope.

cameroncros, 2022-09-17:
@Jaddy4567 Are you happy now? If everything is working, can I close this ticket?

Jaddy4567, 2022-09-18:
Yeah all fixed, Thanks so much for all your help.

closed 2022-09-18 by cameroncros