cameroncros/OctoPrint-DiscordRemote #227: Issues after update

mrquiglz, 2022-09-15:
**Describe the bug**
After updating the plugin via the automatic announcement via octoprint and restarting my server, the plugin is no longer available. It isnt listed anywhere in my plugin manager and errors occur when i attempt to install via URL. I love this plugin as i use it daily. Hoping to see it fixed soon.
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/9571891/plugin_discordremote.log)

*Logs*
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/9571891/plugin_discordremote.log)

**Desktop (please complete the following information):**
 - OS: WIndows 11 PC
 - Browser Firefox

cameroncros, 2022-09-15:
Are you able to do a `pip list` on your octoprint virtual env?

mrquiglz, 2022-09-15:
> Are you able to do a `pip list` on your octoprint virtual env?

I can try.

mrquiglz, 2022-09-16:
This is what i found after running the command. I even reinstalled the server. When launching it command prompt says there is an issue loading plugin DiscordRemote. Im eager to see if i messed something up or if its a compatibility issue.
[image]

MaleNurse, 2022-09-16:
Not sure if it's the same since the above logs contain only a single line, but my issue came down to discord.py [removing EmptyEmbed in v2.0](https://github.com/Rapptz/discord.py/commit/c6ab67420ef7e8a1cd5bc9719bfb84247b72f82c). My logs showed `cannot import name EmptyEmbed from discord.embeds`
When I downgraded discord and discord.py to v1.7.3 via pip everything started working again. Since their pip list shows 2.0.1 for discord (the latest release) I would wager that their problem might be the same as mine.

cameroncros, 2022-09-16:
Urgh, this is getting very confusing. @MaleNurse what version of python are you using? discord.py 2.0 only supports python 3.10+, and I was assuming that most people would be on an earlier version of python?

MaleNurse, 2022-09-16:
@cameroncros yessir, Python 3.10.  Python deprecated and removed the Windows installers for anything prior so I was stuck with it on a new install.

cameroncros, 2022-09-17:
Okay, I have released an update that pins discord.py to 1.7.3. Hopefully it should work for just about everyone, and an update is all that is needed. Please let me know if there are further issues, and I will close this ticket in a week or so

MaleNurse, 2022-09-22:
@mrquiglz working for me! thanks @cameroncros!

closed 2022-09-22 by cameroncros