cameroncros/OctoPrint-DiscordRemote #228: DiscordRemote - not responding on commands through Discord

nabzster, 2022-09-21:
Hi,

I have an Ender 3 Pro with Octoprint and I decided to use the DiscordRemote plugin. I installed everything and I'm getting the bot to type messages in Discord but I can only run the "/" commands in octoprint. If I run it on Octoprint, the bot would respond in discord. If I use my user in Discord and type for example, "/status" it will not respond at all. I've tried setting the user permission with my ID and "*" same with the commands to "*" but have had no luck.

I'm not sure if the BASE URL is correct for my setup. I'm just using the https://[local-ip]/api/plugin/discordremote

The DiscordRemote is always green on octoprint.

Not sure what I should do next, thank you so much for the help.

samuellogan, 2022-09-25:
Have you tried setting rule #1 as follows:
[image]
I didn't make use of the base url and left it blank. Sorry i can't be of any more help lol

gopgop9000, 2022-10-02:
Having the same issue, it detects the message when I use a command, but does nothing with it (from the Debug level logs). No reply or anything, and no further information on the logs.
[image]

gopgop9000, 2022-10-02:
Just looked at closed issues (I really wish that was the default) and it was fixed by the steps in https://github.com/cameroncros/OctoPrint-DiscordRemote/issues/226.

nabzster, 2022-10-03:
Holy crap. It works! Thank you so much.

cameroncros, 2022-10-03:
Sounds like I need to update the doco if this is tripping up lots of people. Will try do that asap

DrToxic, 2023-01-03:
Not sure if still an issue or not. but check gateway intents.

This is how I got DiscordRemote to work.

[image]

closed 2023-01-07 by cameroncros

Ryan526, 2023-03-12:
> Not sure if still an issue or not. but check gateway intents.

Thank you this fixed my issue!