cameroncros/OctoPrint-DiscordRemote #230: Failed to Find File

nick-elzer, 2022-10-12:
Hi! I have successfully installed the bot and have the server up and running. All commands are working fantastically so far, except when trying to access the files on the printer. When I try the /print {filename} or /getfile {filename}, it states that it failed to find the file. I have used /files to ensure that the file I am referring to is in fact on there and that it sees it, and I have tried typing it everyway possible. Any suggestions?

closed 2022-10-12 by nick-elzer

cameroncros, 2022-10-12:
What was the solution if you don't mind me asking?

nick-elzer, 2022-10-12:
I was sleepy brained and apparently "typing it everyway possible" did not include without the brackets lol. Tried it again this morning and it was the first thought, now everything works beautifully. Great plugin!